ramapcsx2/gbs-control #399: WiFi connection won't work at all with multiple ESP's

Kentsfield93, 2022-12-23:
Hey there,

I just finished building my GBS- Control but no matter what I do - I can't connect to the initial wifi "gbscontrol" with the pass "qqqqqqqq".

I didn't change any code, left everything as it is with the current build. I'm using a Wemos D1 mini clone which is using the ESP8266EX chip. I think that this might be an issue but according to all I found using google, it's 100% pin, code and register compatible with other ESP8266 modules.

The first wemos I used, didn't even started the wifi station. I then used another one, which did finally show up on my phone but I can't connect no matter what I do. This also happened with a 3rd and 4th wemos I tried.

Phone no 1 (Google Pixel 7 pro) just says "can't connect" without any details.
Phone no 2 (Oneplus 8 pro) says connection refused/denied by the ESP8266...?
PC with wifi also says can't connect.

I can rule out that Wemos nr 2 to nr 4 are faulty, rudimentary example sketches work just fine and I can connect to it in STA mode. Lastly, I changed the default pw from qqqqqqqq to 12345678, which also didn't work (same results as with the default qqq pw).

At this point I am completely unsure whether this is an esp8266EX problem or not. I know that ESP 2-4 are working, but what else could it be? Sadly, I don't have any other wemos or other "tinker boards" laying around here which I could try, only a load of barebone ESP8266 from the very early days and a load of ESP32 WROOM and WROVER modules, but I don't know if the code is 1:1 compatible with an ESP32.

Besides from the wifi issue, my GBScontrol seems to work just fine. The debug console is pretty verbose and spits out nice facts, video and resultion switches are being picked up correctly and the video output via a cheap HDMI adaptor is gorgeous! Sadly, my OLED and rotary enconder didn't arrive yet, otherwise I could just don't use the web interface at all...

Any help is appreciated! If it is indeed an ESP8266ex specific problem, I think the code might needs to be adjusted as these modules are getting more and more popular as I can see (probably just a cheaper replacement for existing ones - It misses the RF shield completely)

Kentsfield93, 2022-12-24:
Update - I ripped out another Wemos D1 mini with a conventional ESP-12N chip from another project, flashed it and - weirdly enough - It immediately connected to my home wifi and was easily accessible via the web browser...?! I am not sure if there were left over remains from my old project in EEPROM or so, because it was connected to my wifi there, but I never gave the GBS-C code my wifi credentials. It just automagically connected...

I then copied the other code which was initially running on the new old wemos to the new one with the ESP8266EX module and it works flawlessly, but it has worse wifi reception. This means it must be the code not working on the new EX style of ESP8266... Maybe that's something which needs to be looked at. 

Anyway, thanks so much for this splendid piece of work and merry christmas!

ramapcsx2, 2022-12-28:
ESP8266EX? Yeah, chip updates could be an issue :p

RobertoMercadoG, 2023-01-20:
Confirmed this. Not all ESP8266 are made equal. I'll run a test battery in the future and report my results.

WacKEDmaN, 2023-02-06:
can confirm.. just purchased a wemos d1 mini clone with ESP8266EX specifically for GBS-C ... i cant get gbs-c to run on it at all...it flashes ok, but nothing happens... also using an older esp8266ex (on with breakout board).. nothing there either, they dont even spit anything out over serial apart from the boot info..
ive even tryed playing with all the settings in arduino ide, for crystal speeds, flash speeds, dio/qio mode, etc etc...
ive been using esps for close to 10 years now and have never seen an issue like this...

i will try using ESP13 and ESP201 boards... and report back how they go...

WacKEDmaN, 2023-02-09:
esp13 works with debug pin change..
esp201 is only 1MB..so didnt bother trying it..
seems the D1 mini clone i have maybe bad..