ramapcsx2/gbs-control #400: Sync issue with some games

Snesuser, 2023-01-06:
I have recently bought a Gbs control from Aliexpress and Donkey Kong Country 2 for my snes gives the following flickering. Can anyone explain why it happens only to this game and how I can fix it?

https://youtu.be/C53Di69HKbw

I am a newcomer when it comes to upscalers and Rgb.
Thank you!

M-Reimer, 2023-01-31:
I guess it could be pretty difficult to support you in this case. For a self made system you could tell us how you wired your GBS, if you use a sync stripper, which circuit you used between GBS and the RGB out of the SNES.

The units, sold on aliexpress, look pretty nice. Actually interesting that someone went that far and created custom boards just for GBS-Control. That's not just a GBS with some addon board. But without a schematic it is difficult to know if they did everything right.

windship, 2023-03-16:
https://www.retrorgb.com/csync.html

whitezombie2000, 2023-03-25:
I just bought one these GBS controls from China and it works fine with Donkey Kong Country 2. I was using an HD Retrovision component cable. I noticed in your YouTube comments that you fixed the issue by getting a better cable, therefore you should close this issue. Glad you got it working!