ramapcsx2/gbs-control #402: GBS Control producing weird lines across the screen

alextwothousand, 2023-01-10:
My GBS is producing artifacts or weird lines on my screen. Help would be greatly appreciated![image]

PSHarold, 2023-01-20:
I encountered this once. It had something to do with RGB/YCbCr settings both in PS2 and Web UI of gbs-c (there's a "toggle between rgbvh/component" switch in the Web UI). The output formats of my PS2 and gbs-c did not match. Hopefully this is true with your case.

JAOT, 2023-01-20:
This is a ground wire needing to be plugged, or the sync line needing a capacitor or resistor, but it would really help if you told what hardware/cables you are using.