ramapcsx2/gbs-control #409: Support for old IBM PC Resolutions

dekkit, 2023-02-05:
Hi, thanks again for GBS control! 

Ive recently been experimenting with using the control with some old 486 PC video card hardware and for the most part gbs does a pretty ok job with a few work arounds to support most MS DOS resolutions ive thrown at it.

Ive been able to get a stable picture using an XOR Sync combiner from RGB HV to RGBs based on the excellent article here: https://www.hdretrovision.com/blog/2019/10/10/engineering-csync-part-2-falling-short

And then feeding that combined RBGS into gbs w/ gbs control.

The issue im experiencing:

No matter how much I play around with the 'Picture Control' and the move and scale arrow buttons in the webui  - I'm always missing the right most side of the picture or if I continue adjusting/ messing with the image too much it becomes become blurry and sync corrupted.

On further inspection of the video card refresh rates - they are clearly out of spec.

While this helps explain why some modern lcd panels cannot display it properly.

Are there any further settings in gbs control webui 
that I can try to resolve out of range style refresh rates?

This may be tied in to this
Issue
https://github.com/ramapcsx2/gbs-control/issues/188

ramapcsx2, 2023-02-05:
Hey,
I remember that DOS modes were particularly hard to dial in, and I don't think I got them to a very good state at all.
The problems are a 72Hz in some modes, as well as generally "weird" timings that don't work well with the base presets.
I think I concluded that ultimately, the entire sync analyze / dialing in tweaks part of the code wasn't going to work well with DOS modes (and to some extend with all the PC modes).

So your cut off side is probably this problem. What needs to be done is finding a different slot of timings for the ADC and the Display Processor that allow sharp scaling and a good picture. Maybe you can find something workable with the HTotal controls..

dekkit, 2023-02-09:
Thanks for the quick reply!

If I were to dial in this old vga card with a few hacks to the gbs control.ino file.

Is there a code reference / wiki I can read through to understand how the Htotals have been implemented.

Thinking further, on the input side (assuming RGBs )
- where in the code is best to start tweaking Horizontal refresh ranges
- vertical refresh ranges

On the output / scaler side
- the relevant code for tweaking the screen resizing.

A bit of a learning curve for me, but I think if I can progress my understanding - this will help a few folks out.