ramapcsx2/gbs-control #411: Help! Blank screen.

ZeFoxii, 2023-02-09:
My gbs control isn't displaying through hdmi or vga, I'm also new to this and bought the tzt one from AliExpress. I looked and the jumpers are bridged from visual inspection. I'm not sure what to do to get normal composer to work!

ramapcsx2, 2023-02-09:
"Yellow White Red", you mean a standard Composite Video cable, right?
This hardware supports RGB and Component, but there is no decoder in it that can deal with Composite (or S-Video).
You need an RGB cable.

ramapcsx2, 2023-02-09:
As a quick test, you can plug the yellow cable into the green Component port.
If it works, you should get a bad looking black and white picture.

ZeFoxii, 2023-02-09:
Here is the pasteboard link so you can see what I see

https://pasteboard.co/YgajHDJcOD6S.jpg

ZeFoxii, 2023-02-09:
It says no signal but I have no the the vga and hdmi connected, and the av cables connected is there some setting I need to set that isn't clear or is it plug in and it should show up?

ramapcsx2, 2023-02-09:
It seems to be unstable, so it won't work. Maybe a different console will work though. The green port with Composite video is a hack only. It's not meant to work like that :p

ZeFoxii, 2023-02-09:
Like this correct?[image]

ZeFoxii, 2023-02-09:
The blue light no longer shows anything f is plugged in

ZeFoxii, 2023-02-09:
Question what does this mean?[image]

ZeFoxii, 2023-02-09:
I got something f but a bad quality was not what I see

ZeFoxii, 2023-02-09:
Now that I know there is a chance money hasn't been wasted I'm gonna take the time to ask some other questions,

Like what is this and what does it do,I messed with it in my moment of panic.
[image]

And what is this I also messed with this in my moment of panic,

[image]

ramapcsx2, 2023-02-09:
This is entirely different hardware. It seems to be an addon board that has a Composite Decoder.
Please undo everything and plug your yellow cable back into the yellow port that your device has.

ZeFoxii, 2023-02-10:
I have confirmed the board does work with a snes. All I wanted to do was test it to make sure it wasn't broken and I have successfully done so.

ZeFoxii, 2023-02-10:
Apparently the n64 isn't rgb comparable

ZeFoxii, 2023-02-10:
That knob I was turning earlier controls how red or green it is, I thought that was controlled via software, I have a question do you guys have a discord channel? That way if I have any other questions or concerns I can ask not on GitHub?

ramapcsx2, 2023-02-10:
You may find some info on the big retro gaming server if you look around:
https://discord.gg/KghjV79z

The server has a role system, so you may have to adjust your roles.

closed 2023-08-02 by ZeFoxii