ramapcsx2/gbs-control #419: (Question) Would installing GBS-control be possible on those new CGA to HDMI converter ?

Odasan, 2023-02-21:
Hey, as the title says it. There's a recent board i saw popping on aliexpress that does VGA/RGB/CGA/EGA/YUV input to HDMI output, here's the aliexpress link [shorturl.at/jyDEK](url) 
The layout of the board looks close to the gbs 8200 one, so did anyone tried to install gbs control on it ?
Would be curious to try this out, i have a gbs control since 6 month and i'm verry happy with it, a version with a direct HDMI output would be a cool upgrade.

If you have any information hit me up.
Thanks

[image]

israpps, 2023-02-21:
this looks very interesting...

would love to hear it works!

YoImLee, 2023-02-27:
I believe people have gotten these to work in the past, just keep in mind that AFAIK there's no way to get audio out of the HDMI output.

israpps, 2023-02-27:
> I believe people have gotten these to work in the past, just keep in mind that AFAIK there's no way to get audio out of the HDMI output.

if it is one. it will be quite a challenge to solder them XD

finos2, 2023-08-20:
If the board posted [here](https://github.com/ramapcsx2/gbs-control/issues/419#ISSUE-1593821962) works just like the ones with the vga then, for the concerns that @YoImLee expressed about sound, or @israpps about soldering on the hdmi, gave me a idea. 

Hdmi first and foremost carries digital pcm (Iirc) audio, you cannot just soldier stereo analog audio to its pins, what is in my opinion possible though, if we find any part number/pinout of the chip that drives the hdmi possibly we could tap analog audio there, since this is the dac for the system.

YoImLee, 2023-08-21:
People actually have gotten HDMI audio to work with these using a PCM1808 as shown here: https://github.com/ramapcsx2/gbs-control/issues/165

finos2, 2023-08-21:
> People actually have gotten HDMI audio to work with these using a PCM1808 as shown here: https://github.com/ramapcsx2/gbs-control/issues/165

Oh! That's exactly what I had in mind ! Glad it works! 

And the [Component Addition](https://github.com/ramapcsx2/gbs-control/issues/165#issuecomment-1221336768)  
That much more tempting  to try to imlement instead of trying to hunt down a good vga-HDMI to use with the stock GBS8200
BTW how is the lag on the built in converter ?