ramapcsx2/gbs-control #423: lifted pin

Firas-Dhouibi, 2023-02-25:
so here the story
a accidentally grabbed the wire that connects the clock generator and the pin 40 on the trueview5725 chip and i ended up with a lifter pin, i cant soldered it back cuz there is no copper left.
is there a solution or better get another board ?

nyanpasu64, 2023-02-26:
I *think* (am not confident, I didn't design the hardware) that you need to connect the clock generator to the chip's pin 40, and the trace on the PCB is not important. Is the pin still attached to the chip? Is the *trace* or the *pin* broken with "no copper left"?

Firas-Dhouibi, 2023-02-26:
> I _think_ (am not confident, I didn't design the hardware) that you need to connect the clock generator to the chip's pin 40, and the trace on the PCB is not important. Is the pin still attached to the chip? Is the _trace_ or the _pin_ broken with "no copper left"?

yes the pin still attached to the chip, the trace on the board is gone

YoImLee, 2023-02-27:
Lifting pin 40 is not a problem. In fact, the wiki used to recommend lifting it. But since it doesn't connect to anything you don't have to. As long as the pin is still there you're fine.

Firas-Dhouibi, 2023-02-28:
> Lifting pin 40 is not a problem. In fact, the wiki used to recommend lifting it. But since it doesn't connect to anything you don't have to. As long as the pin is still there you're fine.

alright, thanks dude

closed 2023-02-28 by Firas-Dhouibi