ramapcsx2/gbs-control #425: Scaling 240p to 1080p cuts off wifi connection and it can't be detected by my ossc and TV

SNES203, 2023-02-26:
For some reason my Esp8266 loses its connection to my router when upscaling 240p to 1080p. Also my OSSC and TV don't recive a signal. I think that might be a sync problem.
Just in case it is necessary, 
I use a pal wii with a component cable and I play at 60Hz 240p so pal 60Hz.

al3x78-777, 2023-03-12:
I have the exact same issue, I have re-flashed the ESP8266 a number of times and it connects to the wifi and I can access the WebUI but the minute I turn a console on in 1080p the connection drops, the GBS Control still shows up as connected in my router but the GBS-C WebUI is unresponsive. It sometimes after 30 minutes comes back though but rarely.

ramapcsx2, 2023-03-12:
It is the common problem with the WiFi, as first described in the wiki:
https://github.com/ramapcsx2/gbs-control/wiki/various-notes#esp8266-wifi-connection
A proper fix is difficult, but I can recommend that you try to change the ESP8266 position, or try to change the router WiFi channel.

al3x78-777, 2023-03-12:
Thank you so much, I can't believe it was that simple to fix the issue. I've moved the 2.4ghz band to channel 6 (11 didn't work 100%) and here we are changing settings again! I'll check for a more permanent fix but for now it works great! Thank you again!

ramapcsx2, 2023-03-12:
Sure, but I need to disappoint you. It will reappear randomly, as some factors that provoke the problem are variable.
It is a very annoying issue :p

amaiorano, 2023-11-13:
This happens to me as well, where in the web UI I see the no-wifi icon flashing; however, strangely I'm still able to press on buttons and options which seem to go through, despite the UI not updating properly. It's janky, but it works.