ramapcsx2/gbs-control #426: Artifact color emulation

remery66, 2023-02-26:
I would love to see artifact color emulation supported for systems that relied on them for years, such as the old Apples or RS Color Computers. How much coffee would one have to buy to generate some interest? It should be relatively easy to implement, as others have already done it for similar scaler/converters. This is the only feature I really miss on the GBS-Control. Cheers and Thanks!

Nuck-TH, 2023-02-27:
TiVa is not programmable DSP - it is scaler ASIC, so it can't do stuff that is not implemented in its circuity.

remery66, 2023-02-27:
Ah, I understand. Thanks for the reply, and it's still the best bang for
the buck! (We're used to switching over to composite anyway lol)

Cheers!
Bob