ramapcsx2/gbs-control #427: Esp32

ZeFoxii, 2023-02-28:
How likely is it for this project to move to the esp32, the esp8266 is discontinued and the price goes up while the more powerful esp32 gets cheaper. Would there be any benefit at all to run the esp32 in the future instead? Or is it just unnecessary.

Saf98, 2023-11-06:
There is this listing in eBay UK, might be worth checking out if still out of stock.
https://www.ebay.co.uk/itm/115721416079?hash=item1af188ad8f:g:DPgAAOSwkx5kAdY~&amdata=enc%3AAQAIAAAA4EDgWA%2FXshhWiamHaurKCVcdFJT8xI%2FQbbJ6RgSyt7Nz6Xny02BaAtS63Y6eBZTEarWv6OreH%2F3T%2B3bKPRN5KDmh2IHEnrFJd72fhJVNaSiVGquUh2CZaWYzbBoFAqkO0WmG7XwkzNX0%2BKNEi9YfEyKZXmmYVclUtcrs7Yaj%2BQFFOqlTar5OYcQNtaV6xJXA1hKRY8OyCiIeygQsjp5OqShyPnbVIn9X2padm%2BPLWSlVw3MrYcraaLcrwLdmmyksgX2nCPJ5eD1INal8Hw3i71hWyQq2%2FFwkZq5zTj4AXZ5u%7Ctkp%3ABk9SR4TG3dX0Yg

Saf98, 2023-11-06:
Listing title: GBS 8200 Conversion kit boards bundle for GBS Control