ramapcsx2/gbs-control #431: sync skipped sfr wrong: 0.00

TratosFiend, 2023-03-09:
So, I've finally got around to using my GBS control and I'm not able to get any output other than passthrough video. Every  time I choose anything other than passthrough, I get `sync skipped sfr wrong: 0.00`
I've followed the documentation to the T and have done everything right, but I just can't seem to get output on anything other than passthrough. If you can shine any light on what I could do to fix the issue, Please do. I've tried with multiple different consoles (Wii, Gamecube, Dreamcast, Mega Drive, all have the same issue.

TratosFiend, 2023-03-09:
`user command f at settings source 2, custom slot 65, status 11
pal forced 60hz: apply vshift
ADC offset: R:48 G:43 B:44
clock gen reset: 162000000
Output Format: Component
Active FrameTime Lock not necessary with external clock gen installed

preset applied: 1280x960 for NTSC 60Hz 

2345678
Format change: 2 <stable>
pal forced 60hz: apply vshift
ADC offset: R:48 G:43 B:44
clock gen reset: 162000000
Output Format: Component
Active FrameTime Lock not necessary with external clock gen installed

preset applied: 1280x960 for NTSC 60Hz 

sync skipped sfr wrong: 0.00`

ramapcsx2, 2023-03-09:
Seems like an issue with your external clock gen, if that is installed. It could be not working, or only partially working.

TratosFiend, 2023-03-09:
Looks to've been a fault in my clock generator. Installed a new one and now it works fine, though I'm seeing little specks in some parts of the screen. I'm guessing that's possibly to do with power smoothing?

closed 2023-03-11 by TratosFiend