ramapcsx2/gbs-control #433: 8220 board mods. What is the capacitor value?

drawingangnam, 2023-03-10:
[image]

8220 board modification reference photo didn't make sense to me, so I drew a schematic by hand.

Is the schematic in the photo above correct?

If so, what value should the CAPACITOR be?

Any advice would be appreciated.

ramapcsx2, 2023-03-10:
Hey,
I picked a random ferrite I salvaged, and the caps look like some 220nF I had plenty of at the time.
Looking at my soldering work, it would be more important to completely bridge the 0 Ohm R59 resistor to ground.
Even a 0 Ohm resistor has a very small non-zero value.

drawingangnam, 2023-03-10:
Thanks for the lightning advice.
Can I organize the schematic as shown in the photo above?
[image]

ramapcsx2, 2023-03-10:
You need to remove the R58 resistor, of course. But then yep.

drawingangnam, 2023-03-11:
> You need to remove the R58 resistor, of course. But then yep. Edit: Ie, FB replaces R58.

Thanks a Lot!

closed 2023-03-11 by drawingangnam