ramapcsx2/gbs-control #434: Remapping GPIO pins not working

Hercules187, 2023-03-10:
Hi all,

I did post this on the shmups thread but its been a week and my post has still not been moderated, I am trying to remap the SDA and SCL lines, I have tried adding the following lines:

```
#define  SCL = D4
#define SDA = D5
```

But the pins are still on the original settings, I tried used an iteger that represents the GPIO port number but still the same, what am I doing wrong?

Thanks

Steve

ramapcsx2, 2023-03-10:
Hey,
is it even possible to freely pick any pin for the I2C? You may find this in the ESP8266 documentation.

Hercules187, 2023-03-10:
I believe so they are done in software so as long as they aren't hard mapped to something it should be possible...

Valnesh, 2023-03-24:
Changing the pins in this line
`SSD1306Wire display(0x3c, D2, D1); //inits I2C address & pins for OLED`
and adding the pins to this one
`Wire.begin();`
worked for me.

For example:
`SSD1306Wire display(0x3c, 5, 4); //inits I2C address & pins for OLED`
`Wire.begin(5, 4);`

Hercules187, 2023-03-30:
thanks for that Valneshok, I'm not sure I tried updating the wire display but I tried the wire.begins previously as you mentioned it didn't seem to work maybe that was why. The other problem may be that im using gpio2 txd01 and that seems to double the serial output txd0 apparently so I may need to turn off the serial console.