ramapcsx2/gbs-control #439: Installed clockgen and gbs board is not responding

israpps, 2023-03-13:
As title says.
Asked a friend to solder the clockgen (he makes boards for a kinda big factory, so he surely knows what he did)

got it back today, plugged it in and did not get any signal

opened web UI and enabled developer mode to see the log

touching any item displays on log
```
GBS board not responding!
```

any ideas?

will attach pics of the board in a few mins

In the meantime, ill try updating the firmware to the newer builds, as the clockgen library has been updated

israpps, 2023-03-13:
this is a duplicate of #291 buuut. there are no messages there that point into the solution

israpps, 2023-03-13:
[image]
[image]

israpps, 2023-03-14:
@ramapcsx2 any idea?

ramapcsx2, 2023-03-15:
Not really.. The message means that the GBS chip didn't respond on the I2C bus.
It could be power related, signals conflicting (SDA/SCL swap or so), etc.

israpps, 2023-03-15:
[image]

ramapcsx2, 2023-03-15:
Okay, try to remove the clock gen SDA and SCL wires. If it then works, you know that the I2C bus is getting bad with the clock gen on it, and you can look into why that might be.

israpps, 2023-03-15:
> Okay, try to remove the clock gen SDA and SCL wires. If it then works, you know that the I2C bus is getting bad with the clock gen on it, and you can look into why that might be.

could it be swapped SDA/SCL (either by mistake of who installed it or by bad labeled clockgen), I bought it here on argentina, so its 99% not original xd.

israpps, 2023-03-16:
Tested the GBSControl UI clockgen related stuff and it seem to respond

```console
user command f at settings source 10, custom slot 65, status 0
GBS board not responding!
user command s at settings source 0, custom slot 65, status 0
GBS board not responding!
user command r at settings source 5, custom slot 65, status 0
Deinterlacer: Motion Adaptive
user command X at settings source 5, custom slot 65, status 0
ExternalClockGenerator disabled
user command h at settings source 5, custom slot 65, status 0
GBS board not responding!
user command g at settings source 1, custom slot 65, status 0
GBS board not responding!
user command f at settings source 3, custom slot 65, status 0
GBS board not responding!
user command 5 at settings source 0, custom slot 65, status 0
FTL on
Active FrameTime Lock enabled, adjusting external clock gen frequency
user command g at settings source 0, custom slot 65, status 0
GBS board not responding!
user command f at settings source 3, custom slot 65, status 0
GBS board not responding!
```
@ramapcsx2

YoImLee, 2023-03-16:
"FTL on - Active FrameTime Lock enabled, adjusting external clock gen frequency" Is what you want to see on recent versions of gbs control. It is better to leave frametime lock enabled now even with the clock gen installed.

In the first picture, the pin next to pin 40 for the clock generator input is tinned and may be bridged. There might also be a bridge in the second picture at the debug pin. Kind of hard to tell though in both cases, I might be wrong.

israpps, 2023-03-16:
> "FTL on - Active FrameTime Lock enabled, adjusting external clock gen frequency" Is what you want to see on recent versions of gbs control. It is better to leave frametime lock enabled now even with [...]

Issue must be related to clockgen install...

As I used the board without it for months.

And the esp setup hasn't been changed.

israpps, 2023-03-16:
Tested with multimeter...

Clockgen wire that goes into scaler chip is not bridged (measured continuity between CLK0 and the pin next to the one soldered)

Debug pin on GBS chip is not bridged.

To be sure, I checked continuity of SDA/SCL from clockgen pin up to their esp8266 equivalents. Everything ok there

Only possible clue is power.

The esp8266 is taking power from the 3.3v of GBS board

ramapcsx2, 2023-03-17:
Can you please unsolder the 2 clock gen SDA/SCL wires and check what happens then?

israpps, 2023-03-17:
> Can you please unsolder the 2 clock gen SDA/SCL wires and check what happens then?

Same issue

There is a short between r11 and r37... Is that normal?

israpps, 2023-03-17:
Also, made esp8266 take power from PSU

israpps, 2023-03-17:
[image]
I have continuity testing between red and yellow dots

israpps, 2023-03-17:
Disconnected whole clock

GBS board is still dead🙄

israpps, 2023-03-17:
Anyone can help troubleshooting this?

Or should I buy a new GBS board instead

ramapcsx2, 2023-03-17:
So since it's still dead with the I2C wires for the clock gen removed, it means that when installing, you accidentally broke something else. Find that "else" next.
You mention a short between R11 and R37. One side of them would be correct (the Vcc side), the other must not be shorted.
If you probe the 2 sides and both are shorted, this is your problem.

israpps, 2023-03-26:
not sure what else to test

tried everything that came to my mind...
Not sure what else to do!

ser51, 2023-04-12:
I have been successfully using GBS 8220 + ESP8266 + SI5351 for a year. But since yesterday, the device began to give me "GBS board not responding!". Checked all wiring. All OK. Disconnected SI5351, the problem did not go away. My problem was solved only by replacing the Y1 quartz with 27.000. Perhaps someone will help.