ramapcsx2/gbs-control #445: NOTHING from my SNES but tested it with an old laptop and I get an image

prjavelin, 2023-03-25:
This project while it was amazing at first it has become a major headache. PLEASE I need somebody to shed some light into why Im getting these issues. I tried everything I could have with my snes. I have used RGBS cable, SCART cable and SNES2VGA adapter with a VGA cable and get NOTHING. I got the idea yesterday to try with an old pc through VGA to rule out the SNES and lo and behold I got an image with it that way. I still couldnt do some things with the GUI and The WIFI kind of signal on the top left keeps flashing red but I finally had something. Now Whats going on that I cant get Anything out of my SNES?? I added the 100ohm resistor and still get nothing. The monitor or tvs get no signal and eventually turn off but no problem with the signal from the old laptop. 
SNES = old one with multiple chips. not the first one not the one chip. Sorry I cant be more specific. 

[image]

[image]
[image]
[image]
[image]
[image]

4lexei, 2023-04-12:
Do you have a PAL Super NES? My VGA to HDMI adapter only works when I have the console running at 60hz.

I don't think your video cables are the problem. I use cheap aliexpress cables modded by me to remove some annoying noise (shielded audio cables).

nyanpasu64, 2023-05-03:
Not sure if this is the cause of your SNES problems, but did you swap your blue and green wires between the clock gen and the black chip?

And what display and output resolution are you using? Pass Through might have trouble with 15khz input signals (SNES 240p, newer 480i, excluding 480p inputs), and is also known to cause similar Wi-Fi disconnections as you're experiencing (but hopefully doesn't affect scaling, only interferes with the web UI).