ramapcsx2/gbs-control #448: Atari ST highres passthrough

krupkaj, 2023-03-28:
Finally I have GBS with GBS control and OLED. I did little testing with my Atari ST. In color modes works everything OK. But the high res mode doesn't. It is written in other issues, but nevermind. I though that the passthrough mode should work as the monitor can display the Atari high mode output really nice. But unfortunately that doesn't work either. I though that in passthrough mode the signal goes directly from in to out without changes, but apparently GBS do something with it. Please could you check what is wrong? I can provide you more info if need. Here is the screenshot from the debug info.

[image]
[image]

NullArgument, 2023-07-06:
This issue, and issue #190 (https://github.com/ramapcsx2/gbs-control/issues/190) seem very similar. I am also experiencing an inability to display Atari ST mono through my GBS8200. Low and Medium color resolutions work just fine. Mono will briefly flash the correct display, then it appears to lower the resolution, shift the display to the right, one scan line becomes two, and only the top quarter of the screen is displayed. I am using a csync generator instead of separate hsync and vsync signals.

NullArgument, 2023-07-06:
Here is an image of my setup for reference:
[image]

NullArgument, 2023-07-07:
Here is what my mono screen looks like;
[image]