ramapcsx2/gbs-control #450: Video out of sync and terrible quality

uuksu, 2023-04-02:
Hello there.

I have purchased GBS8200 and started modding it with the instructions of the GBS-Control. The board I purchaced seem to be [V5.2 (2022 05 28)](https://uuksu.fi/other/gbs-control/2.jpg). I'm using Wemos D1 Mini that I flashed and installed to the board. My goal is to use the GBS scaler with SCART equipment so I've created a simple adapter as specified in the guide (RGB + S + G connected, 100 ohm resistor between sync and ground). I'm using 5V 3A power source. I output the video with a cheap VGA to HDMI converter I've bought from Aliexpress.

[Here](https://uuksu.fi/other/gbs-control/1.jpg) is a picture of my current setup.

Hardware mods I've applied so far:

- Add 100ohm resistor across sync and ground for RGBs input
- Remove the 3 RGB potentiometers, replace with direct link

Results so far have been pretty terrible. I've tested with a few PAL game consoles (PS1, PS2 and Sega Mega Drive) and all of the have a problems with the image. Here a videos and descriptions of them:

- PS1: [Video is out of sync, colors are wrong](https://uuksu.fi/other/gbs-control/ps1.mp4). TV reports the output resolution to be 800x600 (56p).
- PS2: [Video seems to be mostly in sync, but there is interference on the screen. Colors seem to be somewhat washed out.](https://uuksu.fi/other/gbs-control/ps2.mp4)
- Sega Mega Drive: Video is very dark and barely visible.

I took a log dump from the GBS-control UI after pressing the Restart button on the settings. [Here](https://uuksu.fi/other/gbs-control/log.txt) is the log.

Is there something wrong with my setup? Where I could start debugging these problems?

ramapcsx2, 2023-04-03:
You have a different problem on each console :)
Starting with the Mega Drive, a dark image is usually a sign of the RGB cable not being built to what the console requires.
I would look at that console last.
The PS2 seems to work best so far, but it may have the same sync issue as the PSX.
The PSX is not syncing, most likely there is a problem with the RGB cable, specifically the sync signal. What cable do you use?

Most importantly, you have the yellow button GBS variant, which I warned in length about. It is missing crucial filters on the output stage, as well as having an oscillating voltage regulator.
Start by patching the regulator, to get the system stable electrically at least:
https://github.com/ramapcsx2/gbs-control/wiki/GBS-8200-Variants#yellow-button-v50-or-2017

Good luck! :)

uuksu, 2023-04-03:
Thank you for you answer! :)

I feel pretty stupid about the C11 cap as I read about it for about millions times and yet forget to remove it. Now I've removed it and ordered replacement cap that I'm going to install when it someday arrives.

Unfortunately there seems to be little to none effect to the image quality. I'm using cheap SCART cables ordered from Aliexpress and I'm aware that they're not the best quality, but I have used the same cables with other RGB to component converter and they seem to work nicely there (no sync problems, colors are pretty good).

It seems that actually all my consoles I've tried so far have this sync problem. [I even retried the Mega Drive and earlier it was fine, but now it's having the problem too](https://uuksu.fi/other/gbs-control/megadrive.mp4). [Also my Xbox seems to have exactly the same problem](https://uuksu.fi/other/gbs-control/xbox.mp4). I tried also the PS2 again and who would have guessed, even it has this sync problem now :(

I'm starting to feel that the board has some kind of defect as I doubt that all my cables are bad as they work with other equipment. Has there been any known problems with the recent GBS8200 batches? Is there some variants that are known to just work better?

ramapcsx2, 2023-04-03:
They have been making the same base GBS since forever, and you can always buy one. You need to look for one that looks normal, and has black control buttons.
[image]
Note the upper left video out section has all the components populated. This is an important filter that the "yellow button" variant lacks.

PS2 component cables are inexpensive, and a good way to test a GBS board using the Component Input, which is easier to deal with.

uuksu, 2023-04-03:
I see. So am I reading correctly between the lines that you do not recommend ever buying the yellow button version? By from reading the guide I was under the impression that the original and yellow button variants are equal after the C11 cap removal has been done but I think this might be false.

Is there difference between the versions of the boards? Is there difference between V5, V4 and V3?