ramapcsx2/gbs-control #454: Clockgen uninitialized error

kellenbrew, 2023-04-24:
Having a bit of a weird issue. For context I have only used the GBS with my Xbox over component, and it is displaying output.

I have tried two different Si5351 external clock generators. I have confirmed 3.3v is being supplied to the clockgen board, and continuity to SDA and SCL from clockgen board all the way to the D1 and D2 of the ESP8266. First wiring it as show in the wiki, and also directly to the D1 and D2 pins. 

When I enable frametime lock I get the following output: 
`sync skipped ofr wrong: 28.77`
`user command 5 at settings source 0, custom slot 65, status 1`
`FTL on`
`Active FrameTime Lock enabled, adjusting external clock gen frequency`
`Error: trying to tune external clock frequency while clock frequency uninitialized!`

The last error repeats in the console. I still have a display, albeit with tearing.

Disabling the external clock and turning on FrameTime lock I get what I assume to be the normal output.
`Active FrameTime Lock enabled, disable if display unstable or stays blank!`

No more tearing and over all a great picture. I am still just unsure as to why my clock gen isn't working. Is this a bug, or are there other things I should check?

nyanpasu64, 2023-05-03:
Is your debug pin not properly mounted? https://ramapcsx2.github.io/gbs-control/Wiki/Build-the-Hardware.html#connect-debugpin

kellenbrew, 2023-05-03:
Double checked my debug pin for continuity to D6, all good there. Tried a Wii with component cables and XBox with official MS component cables and same issue. Changed TV's and without anything attached to the VGA output and same error. 

Hooked up a laptop to the VGA input, and enabling the clockgen gives no errors and works fine.

kellenbrew, 2023-05-03:
Sorry I think my words got jumbled in my first message.
Both the Xbox and Wii are softmodded, set to output 480p

Clock gen enabled my monitor reports this:
[image]

Disabling the clock gen and rebooting, not changing the input in any way
[image]

nyanpasu64, 2023-05-03:
Where did you order your clock generators from? Are they outputting half the requested frequency due to a lower frequency crystal?

Do you also get 29hz output on commit 34c816ac4dfb16a503e93145ef2166e4f6d0ad13 (before some clock gen changes)?

kellenbrew, 2023-05-03:
I can try that version. Here is the clock gen I ordered 
https://www.amazon.com/dp/B07X7ZT7KJ?psc=1&ref=ppx_yo2ov_dt_b_product_details

Also for further testing, I connected the output via VGA to a different LCD monitor. Previously I was using a VGA to HDMI adapter.

Clock gen enabled:
-Pass through H: 31khz V 60khz (different than H: 60 V: 60 observed on other monitor with HDMI adapter)
-Any upscaling preset: `sync skipped ofr wrong 28.77`, monitor does not support input from GBS-C

Clock gen disabled:
- Pass through H: 31khz V 60khz
- Any upscaling preset : Works and reports 60 V/H , same as other monitor through HDMI adapter

nyanpasu64, 2023-05-03:
Looks like that board has a 25 MHz crystal like the code expects, so that's most likely not the problem.

If the old commit fails too, did you verify you connected the center pad of CLK0 to pin 40 on the Tvia chip with heatsink? https://ramapcsx2.github.io/gbs-control/Wiki/Si5351-Clock-Generator-install-notes.html

kellenbrew, 2023-05-03:
Using that commit, looking at the serial console over USB I get `no ext clockgen`. Current master branch does detect it. I verified my wiring but I will reflow the connections just to make sure.

kellenbrew, 2023-05-03:
reflowed connections. Confirmed SDA/SCL pin continuity from points on clock gen to points on ESP. Confirmed 3.3v on ground and side of the cap on clockgen.  Same result of not detected with suggested commit.
[image]

kellenbrew, 2023-05-03:
Added some debug code, pass through with horizontal 31khz 60khz horizontal, the field time ticks was 5560856. So the ofr message math checks out with 160000000.00/5560856 . Not sure if the field time ticks are correct though.

nyanpasu64, 2023-05-03:
I'm starting to think you have a clone chip which doesn't communicate properly with the GBS-C (outputting the wrong frequency), and was unrecognized prior to #429. Can you check that the clock gen is detected (but malfunctions) on 6d7e89f5bfdd6cd4c422f57a8be9df99ac8bd5f4 (which adds support for clones) and not on the previous commit 4b20ae47ed6b428019fff6eaf924f9721d6d20ce?

As for fixing the functionality, I'm not sure how to approach that. You could try reverting 766efbf4896b842f464c841d04aa19c0f414f15f then 8f565da9383d2582de5b44370500f09ef37c02d1, and see if your chip outputs the right frequency or not. If that doesn't work, then perhaps the library at https://github.com/pavelmc/Si5351mcu doesn't function correctly with clone chips (it does some trickery to allow for smooth frequency sweeps), though I currently don't know what's wrong as I don't have a clone chip.

kellenbrew, 2023-05-04:
I think you may be right. With the clock gen enabled the GBS board does stop responding after jumping around a few presets until I reboot. I ordered both from the same vendor, bad idea in hind sight. I have one coming directly from adafruit now. 

If it is just the wrong frequency, is there any data I can collect jumping between the commits that may help support this clone for others in the future (short of an oscilloscope)?

Thanks for your help!

nyanpasu64, 2023-05-04:
@smgoldade is your clone clock gen fully functioning with correct output frequency? This issue's reporter appears to have a clone clock gen that's recognized with your patch, but it doesn't work properly with GBS-C's Si5351mcu library.

Interestingly, both the Amazon link (a clone chip, some good non-GBS-C reviews?), and the listing I purchased (https://www.ebay.com/itm/384925804207, worked with unmodified code as of my purchase at 2022-11, no clue now), are labeled Si5351A rather than Si5351. 

Are there perhaps some registers that need to be initialized? Though I'm currently too busy to debug the library code and see what's missing. It's possible it could be made to work with changed initialization, changed `Si5351mcu::setFreq` code, or alternatively the clone chip is outright incompatible with smooth frequency sweeps which external clock gen and especially framesync depends on.

Documents:

- https://cdn-shop.adafruit.com/datasheets/Si5351.pdf
- https://www.skyworksinc.com/-/media/Skyworks/SL/documents/public/application-notes/AN619.pdf

kellenbrew, 2023-05-04:
Just an update 
ofr wrong on all of them
6d7e89f - same error 	(detected)
4b20ae4 - same error	(detected)
766efbf - same error	(detected)
8f565da - same error	(detected)

Possible the supplier could have changed vendors. Honestly the only reason I bought that specific one was because it was linked in the parts list under Voultar's gbs video.

smgoldade, 2023-05-04:
My clone chip is still working. I use it for my NES/SNES. It's possible
indeed there is even more of another variety that have even more registers
misconfigured?

nyanpasu64, 2023-05-04:
I don't know what's going on. If it was a clone chip it would be "no ext clockgen" prior to 6d7e89f5bfdd6cd4c422f57a8be9df99ac8bd5f4 "Merge pull request #429 from smgoldade/robust-clockgen" on `11 Apr 2023`. But you're getting detection with errors on all commits starting from 8f565da9383d2582de5b44370500f09ef37c02d1 "Update Si5351mcu to unmodified upstream 0.7.1 (#435)" `12 Mar 2023` which changes the clockgen *usage* but not setup code, but not on the previous commit? Can you retest on 34c816ac4dfb16a503e93145ef2166e4f6d0ad13 (before #435) and see if the chip is detected or not?

I don't think it's very likely the clock gen's underside is shorting against the TV5725's heatsink?

kellenbrew, 2023-05-04:
34c816a clock gen is not detected. Short is unlikely as the board is sitting in a 3D printed wood PLA holder.

nyanpasu64, 2023-05-04:
So 8f565da9383d2582de5b44370500f09ef37c02d1 has "ext clockgen detected" followed by malfunction, but 34c816ac4dfb16a503e93145ef2166e4f6d0ad13 has "no ext clockgen"? I don't know why that would happen, as the value of `rto->extClockGenDetected` is set strictly before we call any methods on `Si5351mcu Si`. I don't think I can debug further without having the malfunctioning hardware on hand.

kellenbrew, 2023-05-08:
Replaced the clock gen with a genuine one from Adafruit. Now adjust clock correctly with no errors.

closed 2023-05-08 by kellenbrew