ramapcsx2/gbs-control #457: Clockgen question

alucryd, 2023-04-27:
I got a prebuilt gbs control from aliexpress, because I wanted an additional composite input as fallback for when I can't go the component way.

However the clockgen is not installed like your documentation suggests, and sure enough my TV loses sync every other second when I try to activate it. CLK0 is connected to the correct pin to the correct chip, SCL and SDA as well. However instead of using C48, they used an additional wire connecting GND to a pin of the same chip SCL and SDA are connected to, and there is also a wire coming from C14 that seems to go to VIN on the clockgen. I say seems because on the unit I got, it is hanging in the air... Is it safe to reconnect it to VIN? Have you seen this configuration before?

A few pics:
[image]
[image]

israpps, 2023-04-27:
where is the esp8266?

alucryd, 2023-04-27:
It's on a secondary board, they used quick release connectors to assemble everything, here's the one I purchased: https://fr.aliexpress.com/item/1005005477690882.html?spm=a2g0o.productlist.main.75.7f661a75PwEkP7&algo_pvid=0af181bf-a46a-4dfa-b7e0-a0d0b180ee39&algo_exp_id=0af181bf-a46a-4dfa-b7e0-a0d0b180ee39-37&pdp_npi=3%40dis%21EUR%21140.7%2199.9%21%21%21%21%21%40212279a216826070881678992d07dc%2112000033261394288%21sea%21FR%212762705032&curPageLogUid=ZFC1Q2v0zoxk

closed 2023-05-08 by alucryd