ramapcsx2/gbs-control #460: Problem with hsync

Sedu551, 2023-05-06:
Helli there!

First time posting in here and im not english native speaker so please bear with me. Well about week ago i just finished making GBS-C and it was working great but... Everytime when i switched to 1920x1080 there was a horizontal line scrolling on the screen with clock gen already installed. I was sure there was something wrong with clock gen itself or my connections and sure... I connected scl to sda and sda to scl. Well now when everything is how it supposed to be... Nothing has changed ? Everytime when i switch Active frame time lock enabled in settings i get an ERROR then when i restart device i just dont see anything on screen. I have to disable external clock gen then restart gbs and everything works fine expect clock gen has to be disabled. I have log which i hope helps with understanding my problem.
I have:
Gbs8220
Clock gen si5351
Lolin nodemcu v3
Oled addon
Sync stripper
[image]

Restart
user command a at settings source 1, custom slot 67, status 14
<reset>
Activity detected, input: RGB

2345678
Format change: 2 <stable>
ADC offset: R:43 G:43 B:44
HTotal Adjust: -12, source Hz: 49.999, output Hz: 50.002

preset applied: 768x576 for PAL 50Hz 

user command X at settings source 1, custom slot 67, status 14
ExternalClockGenerator enabled
user command 5 at settings source 1, custom slot 67, status 14
FTL on
Active FrameTime Lock enabled, disable if display unstable or stays blank! Method: 1 (vtotal only)
user command i at settings source 1, custom slot 67, status 14
Active FrameTime Lock enabled, disable if display unstable or stays blank! Method: 0 (vtotal + VSST)
user command i at settings source 1, custom slot 67, status 14
Active FrameTime Lock enabled, disable if display unstable or stays blank! Method: 1 (vtotal only)

Restart
user command a at settings source 1, custom slot 67, status 14
<reset>
Activity detected, input: RGB

2345678
Format change: 2 <stable>
ADC offset: R:43 G:43 B:44
clock gen reset: 81000000
HTotal Adjust: -12
Active FrameTime Lock enabled, adjusting external clock gen frequency

preset applied: 768x576 for PAL 50Hz 

sync skipped ofr wrong: 24.00
Error: trying to tune external clock frequency while clock frequency uninitialized!
Error: trying to tune external clock frequency while clock frequency uninitialized!
Error: trying to tune external clock frequency while clock frequency uninitialized!
Error: trying to tune external clock frequency while clock frequency uninitialized!
user command X at settings source 1, custom slot 67, status 14
ExternalClockGenerator disabled
Error: trying to tune external clock frequency while clock frequency uninitialized!
Error: trying to tune external clock frequency while clock frequency uninitialized!

Restart
user command a at settings source 1, custom slot 67, status 14
<reset>
Activity detected, input: RGB

2345678
Format change: 2 <stable>
ADC offset: R:43 G:43 B:44
HTotal Adjust: -12, source Hz: 49.999, output Hz: 50.002
Active FrameTime Lock enabled, disable if display unstable or stays blank! Method: 1 (vtotal only)

preset applied: 768x576 for PAL 50Hz`

Sedu551, 2023-05-06:
Here i have log from GBS-C when PS2 is working and video is displayed without clockgen then i switch clockgen to enable and restart device. After restart there is no video on screen until i disable external clockgen and restart device.
user command X at settings source 1, custom slot 67, status 14
ExternalClockGenerator enabled

Restart
<reset>
Chip ID: 1A 1
<reset>
Activity detected, input: RGB

2345678
Format change: 2 <stable>
ADC offset: R:43 G:42 B:44
clock gen reset: 81000000
HTotal Adjust: -12

preset applied: 768x576 for PAL 50Hz 

sync skipped ofr wrong: 24.00
user command X at settings source 1, custom slot 67, status 14
ExternalClockGenerator disabled

Restart
user command a at settings source 1, custom slot 67, status 14
<reset>
Activity detected, input: RGB

2345678
Format change: 2 <stable>
ADC offset: R:43 G:42 B:44
HTotal Adjust: -12, source Hz: 49.999, output Hz: 50.002

preset applied: 768x576 for PAL 50Hz

nyanpasu64, 2023-05-08:
"sync skipped ofr wrong" reminds me of #454. It's possible you got a bad clock generator. Can you try commit 34c816ac4dfb16a503e93145ef2166e4f6d0ad13 and see if it gets detected or not? Where did you purchase your clock generator board from?

Sedu551, 2023-05-08:
I purchased clockgen here:
[Clock](https://m.pl.aliexpress.com/item/4000012781115.html)
I know that i shouldn't be buying fron ali but ehh it was one massive delivery with gbs8220 too. I just ordered a new one from trusted company i hope it will be good one.
About that commit well I'm not educated enough to understand what i exactly have to do but i suppose you want me to reflash node mcu with that exactly firmware revision ? Well i just did that and everything went to hell heh it's not so simple to get it working. But after 4h looking for tutorials etc. I got it working and everything was the same no new errors i didn't even see new msg "no external clock detected" so i just went back into newest master revision and desoldered clock gen. I will comment again when new one appears. Again like i said Git is new to me and i try my best with my english.

Sedu551, 2023-05-10:
Ok i just soldered a new clock gen and GBS seems to be stable. However I'm getting hmm... Shaky letters and horizontal distorting line slowly running across screen. Here is log.
****
2345678
Format change: 2 <stable>
full height
ADC offset: R:43 G:42 B:43
clock gen reset: 81000000
HTotal Adjust: -7
Active FrameTime Lock enabled, adjusting external clock gen frequency

preset applied: 1920x1080 for PAL 50Hz (set your TV aspect ratio to 16:9!)

.
2Error: trying to tune external clock frequency while clock frequency uninitialized!
source Hz: 49.99869 new out: 49.99869 clock: 80961024 (-38976)
user command 5 at settings source 5, custom slot 65, status 15
FTL off

.h: 431 v: 624 PLL:0 A:4a4a4a S:a7.10.00 H-   I:00 D:1a10 m:2 ht:2553 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-35
h: 431 v: 624 PLL:0 A:4c4c4c S:a7.10.00 H-   I:00 D:2790 m:2 ht:2553 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-35
h: 431 v: 624 PLL:0 A:4e4e4e S:a7.10.00 H-   I:00 D:800e m:2 ht:2553 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-35
h: 431 v: 624 PLL:0 A:505050 S:a7.10.00 H-   I:00 D:0090 m:2 ht:2553 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-35
h: 430 v: 624 PLL:0 A:525252 S:a7.10.00 H-   I:00 D:8f00 m:2 ht:2553 vt: 310 hpw: 182 u:  0 s:ff S:13 W:-36
h: 431 v: 624 PLL:0 A:545454 S:a7.10.00 H-   I:00 D:0f90 m:2 ht:2552 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-35
h: 431 v: 624 PLL:0 A:565656 S:a7.10.00 H-   I:00 D:9000 m:2 ht:2553 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-35
h: 431 v: 624 PLL:0 A:585858 S:a7.10.00 H-   I:00 D:9000 m:2 ht:2553 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-37
h: 430 v: 624 PLL:0 A:5a5a5a S:a7.10.00 H-   I:00 D:1000 m:2 ht:2553 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-37
h: 431 v: 624 PLL:0 A:5c5c5c S:a7.10.00 H-   I:00 D:10c7 m:2 ht:2553 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-36
h: 431 v: 624 PLL:0 A:5e5e5e S:a7.10.00 H-   I:00 D:4c0f m:2 ht:2553 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-39
h: 431 v: 624 PLL:0 A:606060 S:a7.10.00 H-   I:00 D:8e80 m:2 ht:2553 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-36
h: 431 v: 624 PLL:0 A:626262 S:a7.10.00 H-   I:00 D:d58e m:2 ht:2552 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-37
h: 431 v: 624 PLL:0 A:646464 S:a7.10.00 H-   I:00 D:0f00 m:2 ht:2554 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-37
h: 430 v: 624 PLL:0 A:666666 S:a7.10.00 H-   I:00 D:bb93 m:2 ht:2553 vt: 310 hpw: 182 u:  0 s:ff S:13 W:-34
h: 430 v: 624 PLL:0 A:686868 S:a7.10.00 H-   I:00 D:8087 m:2 ht:2553 vt: 310 hpw: 182 u:  0 s:ff S:13 W:-36

. (debug) ign. length: 0x66
. (debug) ign. length: 0x67
user command 5 at settings source 5, custom slot 65, status 15
FTL on
Active FrameTime Lock enabled, adjusting external clock gen frequency
user command 5 at settings source 5, custom slot 65, status 15
FTL off
h: 431 v: 624 PLL:0 A:6a6a6a S:a7.10.00 H-   I:00 D:f1a2 m:2 ht:2553 vt: 310 hpw: 181 u:  0 s:ff S:13 W:-36

nyanpasu64, 2023-05-14:
Do you still get "Shaky letters and horizontal distorting line slowly running across screen"? If not, how did you fix it? (I did less testing of frametime lock with external clockgen, and only used 240p Test Suite's PAL50 since I don't have any PAL games or consoles.)

Sedu551, 2023-05-14:
Yeah i still have shaky letters it looks strange and only happens with 1920x1080 resolution. Amiga 500 doesnt have that issue. PAL PS2 and PS1 only, at least these i can confirm for sure. Horizontal line is still there but now it's not moving with every gbs restart starts to move then about 5-10sec later it stops. I just use GBS as it is eventually i switch to 720p and everything is fine. I have to test it with Wii but i have to make RGB cable first.

nyanpasu64, 2023-05-14:
What happens if you enable Developer Mode and click "Cycle SDRAM" a few times?

> Horizontal line is still there but now it's not moving with every gbs restart starts to move then about 5-10sec later it stops.

Can you open framesync.h, uncomment `// #define FRAMESYNC_DEBUG`, reupload, and show me the debug logs 10 seconds after the line stops moving?

Sedu551, 2023-05-14:
To be honest i think i found shaky letters issue. Everytime when im connecting to gbs via wifi(it's connected to my home wifi) its getting much much worse but only at the time when im trying to connect to gbs. So i think i should make some shield around nodemcu ? About cycling sdram nothing happens. Screen only flickers for half second and later everything is the same. Here is the log
user command l at settings source 2, custom slot 65, status 15
SDRAM clock: 144Mhz
user command l at settings source 2, custom slot 65, status 15
SDRAM clock: 162Mhz
user command l at settings source 2, custom slot 65, status 15
SDRAM clock: 185Mhz
user command l at settings source 2, custom slot 65, status 15
SDRAM clock: 108Mhz
user command l at settings source 2, custom slot 65, status 15
SDRAM clock: Feedback clock
user command l at settings source 2, custom slot 65, status 15
SDRAM clock: 129Mhz
user command l at settings source 2, custom slot 65, status 15
SDRAM clock: 144Mhz
user command l at settings source 2, custom slot 65, status 15
SDRAM clock: 162Mhz
user command l at settings source 2, custom slot 65, status 15
SDRAM clock: 185Mhz
About reuploading is there any fast way of doing it or i have to do that via arduino ide ? With formatting node mcu first then reuploading modified code ?

nyanpasu64, 2023-05-14:
Wi-Fi lmao... yeah video and wifi tend to have interference issues.

> About reuploading is there any fast way of doing it or i have to do that via arduino ide ? With formatting node mcu first then reuploading modified code ?

Yes through Arduino IDE, but you don't have to clear flash when reuploading. In menu Tools -> Erase Flash, pick "Only Sketch" before uploading.

LouisBurdett, 2025-05-22:
Hi OP,
Did you get a happy resolution to your issue?
I see we both have a similar log output that looks like an error. Your post is the closest I've seen you my issue:

`2Error: trying to tune external clock frequency while clock frequency uninitialized!
source Hz: 49.99869 new out: 49.99869 clock: 80961024 (-38976)`

Odd that the log seems to show a frequency correction but still says "Error: trying to change clock frequency..." Does that mean it is, or isn't working?

nyanpasu64, 2025-05-22:
it might be working, but i haven't used the gbs-c in months or touched it for years so i don't recall.

i seem to recall that error comes from a race condition in the code where when i introduced closed-loop clock targeting, it can randomly attempt to run before initialized. and it's only a problem if it keeps happening indefinitely. i couldn't find a way to make it only run when ready, besides ignoring the error if it happened (but that would mask a failure to initialize the clock). one possibility is ignoring it *the first time* only?

the gbs-c initialization state transitions confused me then and i understand even less now.