ramapcsx2/gbs-control #461: random black screen in passthrough mode

nyanpasu64, 2023-05-11:
discord, 04/15/2023:

gbs-c in custom preset -> passthrough, playing nintendont, went black for a second and restored
unsure if crt or gbs-c failed
no message in logs (computer was awake but locked, unsure if logs were making it to the browser properly)

(wii, monkey ball 2 at 480p)

also wii, mkwii at 480p

## possible causes

- got issues with both powered usb hub, and moto usb-a fast charger (pre-PD), both >3 amps

- #459 is unlikely; (2023-05-10) in homebrew channel, i might've just seen my wii lose sync out the corner of my eye, **with auto gain off**.

- from #446? maybe.

- the behavior of the black screen is identical to briefly disconnecting the input signal for short enough time the esp doesn't notice a sync loss and print a dot. idk the cause.

	- https://discord.com/channels/1083378130716209194/1083378131332763750/1098477338473283594
	- i wonder if the y-splitter for recording introduces enough smearing that the gbs-c doesn't register sync wrong anymore, or if the issue just didn't manifest.
	- still waiting on sog capacitors to ship (tomorrow?)
	- i know there's someone on GitHub issues https://github.com/ramapcsx2/gbs-control/issues/449#issuecomment-1506594666 claiming they had to do the sog capacitor mod to avoid signal loss, on the 240p thread, but "xbox connected with component cable", I'm on 480p

nyanpasu64, 2023-05-20:
grrrrr
i swapped the SoG caps and *still* got a dropout on skyward sword
i suspect the issue this time is not related to sog, despite the symptoms matching
but perhaps caused by custom passthrough (which was not working at all prior to my recent changes)

nyanpasu64, 2023-05-24:
I got a recording of the output. The screen flashed black for part of a frame, but didn't lose sync persistently.

[black screen glitch.zip](https://github.com/ramapcsx2/gbs-control/files/11558542/black.screen.glitch.zip)

(this particular glitch didn't cross a vsync pulse so it didn't make my monitor go black for half a second)
to me it looks like the video scaler (GBS-C, TV5725) briefly mistook the 31.5khz 480p signal for a 63khz or so signal, then went black, then started displaying 63khz video again
and the next frame, output a 31khz signal again
unfortunately i did not capture hsync, but the video display signal looks quite like 63khz to me

[image]

[image]

- [ ] In passthrough mode, how does the TV5725 know the incoming video signal's hsync rate?
- [ ] what happens with sync watcher off and switching between 15 and 31khz inputs (in 240p test suite)? does it switch output hsync rate anyway?

nyanpasu64, 2023-05-25:
i have some interesting observations:

- TTYD appears to have a 477-line vertical active period.
- the glitch appeared after my partner's attack animation finished, as the status bars were moving down from the top of the screen to their final position (they moved 1px in 2 frames), as the enemies were attacking (since no ring menu is visible).
- when the image went black, for a *full frame* parts of the image were *completely missing* and indistinguishable from blanking period, I saw *no edges* between vertical porch and active, the entire GBS-C output appeared blanked (outside of 2D HUD elements).

do I have a defective Wii where the GPU shits its pants randomly for a single frame, and fails to render 3D graphics, and sometimes outputs bad sync as well?!

another possibility is that the TTYD black frame and the sync loss are separate bugs.

ramapcsx2, 2023-05-25:
Did you just oscope your video signal with your sound card, or what is that .flac file? xD
So some points:
If you have passthrough configured, the entire operation is a pure hardware chain.
The input gets ADC'd, along with the syncs, and this is used unmodified for the output (hence passthrough. it passes through all the processing blocks without them doing anything).
There are still some configurable elements, especially on the syncs, and please look up what the code configures to get an idea.
I think part of the segment 5 syncs registered are used.

nyanpasu64, 2023-05-26:
Not directly related to this bug, but I found that when switching to fixed passthrough from a reboot or scaling preset, it takes >1 second longer than normal, because `TEST_BUS_2F` returns all zeros and when the program tries to run `optimizeSogLevel`, it always fails at `outer test failed syncGoodCounter: 0`. This is despite `TEST_BUS_SEL = 10, TEST_BUS_SP_SEL = 15` like expected.

If I try again, switching from passthrough to fixed passthrough, it runs `rto->thisSourceMaxLevelSOG = rto->currentLevelSOG = 14;` and the test bus works as-is and `setAndUpdateSogLevel` is never called. (Unless you load preset right after powering off the Wii, then it fails to optimize SOG again.)

Why does this happen?

Separately. perhaps when restoring a passthrough preset, instead of reading all registers I'd just read ADC gain (and offset?) alone, then apply passthrough as usual. Though it may trigger the above mishap as well.

Also I tried cycling `SOG Level--` from 16 through 0, in both fixed and custom passthrough, and saw no immediate sync glitches on monitor or Audacity (only the screen moving horizontally slightly), the output hsync and vsync waveforms looked perfectly periodic. I have not been able to reproduce a glitch on my LCD monitor; I'm ordering a VGA distribution amp to power both my CRT and LCD (and additionally probe the signal like before), since I'm starting to suspect my CRT fails to sync or interprets a marginal h/v sync as losing sync (while my LCD is more lenient?).

Is the GBS-C's output signal in-spec? I found the GBS-C outputs 2 line duration vsync in 480p passthrough mode (matching DMT 480p), but the vertical position is off by 3 lines. (the funny numbers in `setOutModeHdBypass`) I don't have the horizontal precision to probe horizontal timings.

`GBS::HD_HB_ST::write(0x878); // 1_3B`
`GBS::HD_HS_SP::write(0x864); // 1_41`
- [ ] are you pulling hsync low *before* hblank begins? this can't possibly be a legal modeline.

I might also look into high-resolution video capture if I buy a https://github.com/happycube/cxadc-linux3 PCIe card.

Also horrifyingly the Wi-Fi server can process `"/gbs/restore-filters"` *during* `optimizeSogLevel`'s call stack.

ramapcsx2, 2023-05-26:
Nothing on the output is much in spec, I can tell you that :)
SoG level is the detection threshold of the sync portion, that after the analog input circuit (the SoG caps!).
It is required to be tuned in many situations (source consoles), but not all of them. These sync pulses vary wildly between consoles.

ramapcsx2, 2023-05-26:
"are you pulling hsync low before hblank begins?" << please make sure you understand this fully. There are settings called "hb" that are not actually the output hblank period, but something else on the scaling chain, etc.

nyanpasu64, 2023-05-26:
- `HD_HB_ST` is set in `setOutModeHdBypass` which doesn't perform scaling.
- When I decreased `HD_HB_ST`, the right of my screen was cut off.
- The register guide, section `HD_BYPS` says `HD_HB_ST` = "Horizontal blank start position", and "Generate horizontal blank to select programmed data." (whatever that means).

ramapcsx2, 2023-05-26:
Please remind me which segment HD_HB_ST is in, but
"Generate horizontal blank to select programmed data."
sounds like it works on the input formatter, preparing the image in digital format for processing later.
If this is the case, eventhough it's called "horizontal blank", it doesn't have to do with any syncs or timings, it's merely a mask on the digital data.

nyanpasu64, 2023-05-26:
It's in the HD Bypass section:

[image]

In any case I'm working on changing the sync pulse positions and sizes to match VGA (my [auto-modeline](https://codeberg.org/nyanpasu64/auto-modeline) program based off [tomverbeure](https://tomverbeure.github.io/video_timings_calculator)'s timings database), in the hopes that this will prevent my CRT from losing sync.

```
> auto-modeline print dmt 640 480 60
"auto-dmt-640-480-60"  25.2  640 656 752 800  480 490 492 525  -HSync -VSync
# (752 - 656) / 800 = 0.12
```

> Did you just oscope your video signal with your sound card, or what is that .flac file? xD

yep lmao. I'm blessed with a 192khz motherboard with practically no detectable aliasing in mdfourier or recording sync signals with harmonics beyond Nyquist, I assume because it has an oversampled frontend. (I didn't try recording video or hsync signals wholly beyond Nyquist, then checking the noise floor.) A real DSO would be better of course, but this is sufficient for testing the positions of vsync pulses and the regularity (but not contents) of hsync pulses.

## Vertical timings

VGA DMT 640x480 vsync `480 490 492 525` has two lines of vsync.

Probing the GBS-C's passthrough signal with my audio interface, I found that the vsync pulse is 2 lines long, but 3 lines late relative to 480-line-tall video content (for example NIntendont's menu).

To fix this, I had to change:

```cpp
setCsVsStart(525 - 5);
setCsVsStop(525 - 3);
```

- [ ] Is it a good idea to send a vsync pulse at the end of each output frame, rather than the beginning?
- [ ] Will displays get confused if they see the first frame of video before a vsync?
- Will this handle incorrect input vsync detection better or worse than the original code? (I suspect that if the GBS-C detects sync incorectly in the input, you've already lost all hope of the display maintaining sync.)

## Horizontal timings

VGA DMT 640x480 hsync `640 656 752 800` is low for 0.12 scanlines:

```
(752 - 656) / 800 = 0.12
```

I calculated that with the GBS-C in 480p passthrough, the horizontal sync line is low for 0.09 scanlines:

```cpp
GBS::PLLAD_MD::write(2345); // 2326 looks "better" on my LCD but 2345 looks just correct on scope
GBS::HD_HS_ST::write(0x10);  // 1_3F
GBS::HD_HS_SP::write(0x864); // 1_41
// 1 + (0x10 - 0x864) / 2345 = 0.09083155650
```

It's possible that this is preventing my CRT from picking up sync. But then again, CVT 640x480 has a 0.08 scanline long hsync pulse (but at a lower hsync rate, so each pulse is longer), and works fine (I did not verify it doesn't lose sync every few hours):

```
> auto-modeline print cvt 640 480 60
"auto-cvt-640-480-60"  23.75  640 656 720 800  480 483 487 500  -HSync +VSync

(720 - 656) / 800
```

`gtf 640 480 60` has the same horizontal timings as CVT, but I did not test that it works on my display. Interestingly *both* CVT and GTF 640x480@60 have a hsync rate *below* 30 kHz, but my monitor doesn't complain. I hope I won't burn it out running it at that.

In any case I prefer to leave hsync at 0.12 scanlines long (leaving the low pulse time unchanged, increasing the high pulse-end time), since it centers the screen the same way my computer's video output does.

----

Do you know why `optimizeSogLevel` fails when loading a passthrough preset for the first time? Is `setOutModeHdBypass -> doPostPresetLoadSteps` checking for test bus, before `setOutModeHdBypass` finishes initializing the registers required for sync to work? And applying bypass twice, or loading a complete set of bypass registers before `setOutModeHdBypass`, prevents this from happening? (I could test this by calling `optimizeSogLevel` repeatedly at different points and seeing when it works.)

I still don't know why my display was cutting out. And I don't know if increasing hsync pulse width was the fix or not, and it will take hours to even attempt to rule it out. Perhaps I could decrease hsync pulse width and see if it reproduces the issue quickly (meaning the old width was marginal)?

ramapcsx2, 2023-05-26:
Because of time, as always, I really cannot reply to everything :p 
I want to convey to you though that you seem to think of the video specifications, modelines and such, way too strict.
Vsync / Hsync are a continous stream in an analog line. The displays and scalers care about some relationships between them, but they have only a general concept of where the video should be, versus blanking.
This is to say, you are free to move Vsync around, make it longer, or shorter (not too short), and it will be "legal" on all devices still.
The specifications and modelines are there as a guideline, and if you manage to follow them to the letter, your chances of an instantly well positioned picture increase, but there is no further benefit.

nyanpasu64, 2023-06-01:
I've found 3 different issues:

- One time the GBS-C failed to emit a vsync pulse, causing the entire screen to flash black or something for a frame.
- When I touch the CRT screen (or move my rolling chair?), the static shocks cause the TV5725 to produce an erroneous extra vsync pulse, making the screen jitter vertically for a frame.
- Randomly during gameplay (three times so far today), the sync processor will output hsync pulses at 4/3 the correct input hsync rate (31 KHz), for a few scanlines before the output rate decreases below 31 the input rate, then increases to the correct number again. I'm not sure what causes this. I've been getting this issue so far in "save and load custom passthrough" mode with auto gain on or off, not sure about fixed passthrough or custom/fixed scaling.

The GBS-C firmware is in `setAndUpdateSogLevel(14)` mode. I'm not sure if changing this value would affect it, or increasing or decreasing the voltage threshold would change the likelihood of sync errors.

Recordings at [hsync rate 4div3.zip](https://github.com/ramapcsx2/gbs-control/files/11630573/hsync.rate.4div3.zip). I suggest viewing in Audacity at around 1.2 to 1.3 seconds, in spectrogram view with length 32. Unfortunately I do not have high-resolution captures of input luma.

Screenshot of `output 2023-06-01 15:18:11`, showing output hsync pulses at 4/3 the input frequency:

[image]

ramapcsx2, 2023-06-02:
Okay, you want to look at every possible option that is active in passthrough mode. That would be all the section 5 sync processor registers, as well as the HD bypass stuff at least.
For HD bypass, all I know is that this is digital data processing, but it may have some effect on syncs.
The sync processor has all these "SoG" and window parameters, and I was never able to fully map out what everything does, or whether it is active indeed (or when it is active, whether it only applies to some modes and not others).
You could try to compare some segment 5 dumps and maybe try a few different combinations of those filter settings. Maybe something works :)

nyanpasu64, 2023-06-02:
i think the problem is that custom passthrough just gets _so many_ registers wrong compared to fixed passthrough, so many registers different that it's a pain to even list them all 
honestly i don't want to go through the trouble of fixing it anymore

[image]

```cpp
const char HEX_DIGITS[] = "0123456789ABCDEF";

void print_hex(uint8_t val) {
    SerialM.print(HEX_DIGITS[(val >> 4) & 0xf]);
    SerialM.print(HEX_DIGITS[val & 0xf]);
}

void print_page(uint8_t page, size_t begin = 0, size_t end = 256) {
    // SerialM.printf("Page %d:\n", page);

    for (size_t row = begin; row < end; row += 16) {
        SerialM.printf("S%c_", HEX_DIGITS[page]);
        print_hex(row);
        SerialM.print(": ");

        for (size_t col = 0; col < 16; col += 1) {
            auto addr = row + col;

            uint8_t val;
            GBS::read(page, addr, &val, 1);
            print_hex(val);
            SerialM.print(' ');
        }
        SerialM.println();
    }
    SerialM.println();
}

void print_regs() {
    SerialM.println("       _0 _1 _2 _3 _4 _5 _6 _7 _8 _9 _A _B _C _D _E _F ");
    SerialM.println("------ -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- ");

    print_page(0, 0x40, 0x60);
    print_page(1, 0x00, 0x90);
    print_page(5, 0x00, 0x70);
}
```

ramapcsx2, 2023-06-02:
I don't think it's about getting them wrong, compared to some other "reference".
It's more about figuring out whether any of the options have an effect or not, and if one does, picking the best value.

ramapcsx2, 2023-06-02:
> IF_HSYNC_RST
is a digital video data "reset" signal, iirc it meant where a line ends, so modifying it might make more or less of your image show. It shouldn't do actual sync stuff. They just used similar terms here :p 

> PLL_IS
you mostly want to leave the core system parameters like these alone, especially any dividers / multipliers of clocks.
But there could be a PLL voltage register somewhere, that's worth taking a look at.

nyanpasu64, 2023-06-02:
## Fixing loading custom passthrough

I've locally modified the GBS-C source code to include a `Context const& ctx` parameter to `doPostPresetLoadSteps` and its callees, and modified all functions called during preset loading to consult `ctx.isCustomPreset` instead of `rto->isCustomPreset`. This allows `setOutModeHdBypass` to call `doPostPresetLoadSteps` and force it to act like it's loading a fixed preset, even if you're applying a passthrough resolution.

With this change in place, saving and loading a passthrough preset seems to produce practically no change to registers. The only ~~image-related registers I print (https://github.com/ramapcsx2/gbs-control/issues/461#issuecomment-1573231013)~~ parts of the full register dump (serial `d`) to change are:

- `s1_2b GBS_PRESET_CUSTOM` (assumed to be unused)
- `s1_7f MD_HD1250P_CNTRL` (hopefully unimportant? idk, it's special-cased in `writeProgramArrayNew`, and if this is a bug then it might break 24khz *scaling* rather than passthrough mode)
- sometimes `s5_19 PA_SP_S` (recomputed heuristically by `optimizePhaseSP()` every run)
	- `optimizePhaseSP()` seems to be called at the end of *every* `setOutModeHdBypass()` call.
	- [ ] Could `PA_PLLAD phase control` affect hsync output frequency?

For posterity, here is the file containing the bad saved passthrough mode which loses sync (dunno if it still breaks with my load-time code changes): [gbs-control.backup-1685073472398.bin.zip](https://github.com/ramapcsx2/gbs-control/files/11639544/gbs-control.backup-1685073472398.bin.zip)

## Stale registers in fixed passthrough

Concerningly, I found that fixed (not custom) passthrough leaves many registers unchanged from the previous reboot (probably any call of `setResetParameters()`) or fixed scaling preset. I don't know if any of these stale registers (which get saved/loaded through custom preset slots) can affect the outcome sync stability, or they're harmless and the sync bug arises because `doPostPresetLoadSteps` wasn't called in fixed mode.

(The existing `dumpRegisters()` or 'd' prints more register ranges than my hexdump did, including OSD and VDS_PROC.)

- [ ] To investigate, I could write a program which corrupts these inconsistent registers one by one in bypass mode, and sees which ones damage the bypass image.
	- However, finding which registers cause hsync frequency glitches hours apart is slow and tedious; it requires first excluding the registers that destroy the image altogether, then verifying that I get rare sync glitches with the remaining registers corrupted, then slowly bisecting for the register(s) at fault.
- [ ] Alternatively, when applying fixed passthrough, I could somehow reset *all* registers to a known-good state, eg. by calling a subset of `setResetParameters()`, presumably including `loadPresetMdSection()` and `PLL_ADS` but ideally not resetting `serialCommand` and `userCommand`.
	- This reduces the possibility of behavior divergence, but I still won't know which registers matter or not.

Questions:

- [ ] Is the INPUT_FORMATTER segment of registers ignored altogether in HD Bypass? VDS_PROC? OSD probably?
	- MEMORY CONTROLLER and "CAPTURE & PLAYBACK" is mostly blank/zeros when rebooting in bypass mode, but the image still works.
- [ ] Is S0_40, 48, and 4F important? Does S5_1F bit 6 = DEC_TEST_SEL affect whether auto gain etc. work or not?

## Fixing `optimizeSogLevel` without applying fixed passthrough twice

As mentioned at https://github.com/ramapcsx2/gbs-control/issues/461#issuecomment-1563895071, `optimizeSogLevel` fails the first time you apply fixed passthrough after a scaling preset.

- [ ] I could run `optimizeSogLevel` (or a 50ms delay followed by `GBS::TEST_BUS_2F::read()`) multiple times in fixed passthrough, and find what position is the first one to succeed.
- [ ] Then I could move the function call there (and pray it doesn't cause side effects).

nyanpasu64, 2023-06-03:
Bad news: with a custom passthrough preset, with a register dump *indistinguishable* from "reboot to passthrough", I got *another* hsync frequency excursion.

I suspect that I got three excursions this time (two in quick succession, followed by a third frequency jump while the PLL was dipping the frequency to fix the signal phase after the first two, which exposes the third harmonic).

[image]

And by the same logic, my screenshot in https://github.com/ramapcsx2/gbs-control/issues/461#issuecomment-1572921651 actually looks like two excursions in quick succession.

- [ ] Possibly the issue isn't that I set registers wrong, but that I wrote too many registers, or registers I shouldn't have written, which left the unit in an unstable state even after I called `setOutModeHdBypass`.
	- In support of this, I get black screens in my custom bypass preset, saved from a fixed bypass applied twice after reboot. Previously I also got them in my old custom bypass preset (loaded with an older firmware), but on my edited firmware, the registers are identical to a fixed bypass applied after scaling to 960p. This means that it's likely *not* the stale registers causing you to lose sync, since I lose sync in both general categories of stale registers?
		- A third category of registers is if you reboot (zeroes out MEMORY CONTROLLER), scale (initializes memory page), setResetParameters (resetting some but not all parameters set by scaling, I assume it doesn't zero out memory page?), *then* enable passthrough (doesn't write to memory page). I didn't check if that loses sync.
	- However, I find it hard to believe that incorrect chip setup causes *random* loss of sync *hours* later in the middle of gameplay.
- [ ] Does fixed bypass lose sync too, if you wait long enough? IDK.
- [ ] Is my chip overheating? I have the clockgen installed on top of the TV5725's heatsink, but I see a lot of people do that. I got hsync excursions with my GBS-C plastic shell open too. Though the heatsink does feel painful to touch for too long.
- [ ] Is the silicon itself failing, from damage, heat, and/or overvolting and overclocking compared to a stock GBS8200 board? It's not out of the question, considering #428 is a report of inexplicable time-based signal loss as well.

Is fixed passthrough susceptible? I don't know. I don't know if "reboot to passthrough" has the issue, or "reboot, scaling, setResetParameters, passthrough" or "reboot, scaling, passthrough".

nyanpasu64, 2023-06-03:
Looking at the PDF (3.5.2 Loop Filter, 3.5.3 Register Setting Rules), I think a possible issue is that `PLLAD_ICP` was set too high.

- I suspect that this was causing the input PLL (and output hsync frequency) to jump to 4/3 of the input clock. The PLL frequency needed to cause this is 31.5 kHz * 4/3 * 2345 = 98.5 MHz, which is slightly above the maximum 88 MHz of KS=1. This is plausibly explained by the overvolted PLL running slightly out of the stable frequency range.
- In support of this theory, I found that turning off `PLLAD_FS` causes horizontal sync instability, as does putting my thumb on the PCB components near pins 48-49 of the TV5725 (PLLAD pins connected to external oscillator components).
- It's plausible a malfunctioning PLL causes frequency jumps as well, and the smoothly drifting output frequency after a frequency jump suggests the PLL is trying to realign with input hsync pulses by changing frequency without introducing phase discontinuities.
	- This doesn't explain the scrambled signal and missing hsync pulses after a vblank (see `Does fixed bypass lose sync too` above)? IDK.

> [image]

The current code turns FS on, and the PDF agrees with this. Turning it off causes horizontal image instability.

> [image]

But `setOutModeHdBypass` with 480p input sets ICP is higher than the formula suggests (assuming `M = PLLAD_MD + 1`):

> [image]

- [ ] Of course, my calculations would be incorrect if the GBS8200 PCB uses a loop filter capacitance (in series with a 3K resistor between pins 48 and 49) different from 33 nF. I don't know what component it uses, didn't study the traces in this region of the PCB, and the components are quite crowded making it hard to study.

I've tried editing the source code to change `PLLAD_ICP` from 5 to 4 (or 3), and I get a stable image. I've ran it at value 3 for an hour or so without image dropouts. However, an hour is not long enough to conclusively say that the issue has gone away (especially since I observed it would occur multiple times an hour on some play sessions, and never on other ones, possibly due to temperature?). And I want to take a break from binging TTYD...

ramapcsx2, 2023-06-04:
Well, yeah. :)

nyanpasu64, 2023-06-04:
Another possibility is that the issue showed up in passthrough testing, because I was routing the _input_ luma signal through a Y-splitter (near my computer's >1kohm audio jack) to the GBS-C, to look for glitches in the input signal, and it also disturbed the luma sync signals. Then when I didn't find glitches in the input signal, I routed the VGA *output* blue wire (fed through an Extron VGA amp) into the audio jack, at the same time I decreased `PLLAD_ICP` and tried running further tests, and the *routing* changes stopped the sync glitches from happening as frequently.

- Every recording of a sync glitch I have (over the last few days) has input luma on the left channel.
	- I haven't caught a single sync glitch since switching to recording output blue; this could be caused by removing the input splitter/tap, or by decreasing `PLLAD_ICP`, or it could be a coincidence and the bug is still present but hasn't shown up yet.
- Never trust a thing I do; I'm basically a magnet for cursed technology bugs.
- When capturing input luma, why do I place the Y-splitter near the computer's high-impedance audio capture jack, rather than the 75Ω-terminated GBS-C?
	- I found that this produces negligible visual artifacts (aside from luma/sync being delayed relative to chroma), whereas putting the Y-splitter in the middle of the video cable or at the GBS-C produced noticeable horizontal luma smearing.

I've left Homebrew Channel idling for 25 minutes or so, while capturing input luma (replicating the setup I had when I saw the most sync glitches), and haven't *noticed* a sync glitch so far. Fingers crossed.

ramapcsx2, 2023-06-04:
It's not a good idea to introduce things like the splitter, when you want to debug such problems.
They're already quite random and rare, so you'd want to avoid any more of these analog problem sources :p 

PLLAD_ICP: PLL for the Analog to Digital section, current Charge Pump level.
It controls the tightness of the PLL response, and has to be tuned for each situation it has to serve.
I've tried tuning it based on what seemed to work best:
If you set it too low, some artefacts start to appear. If it's too high, it looses all control.
I've set it a few notches above the artifact level. Note: This has to be done for each source / operating mode. I think the source code tells you as much.

nyanpasu64, 2023-06-04:
I just got an incident of frequency spike and sync loss, at `setAndUpdateSogLevel(14)`, `PLLAD_ICP = 3`, and no splitter in the signal path from Wii to GBS-C. Hopefully my hi-res capture card will show up over the next few days.

I looked up `separ` and `seper` in the programming guide and didn't find much useful. Can sync glitches be identified using interrupts (chapter 14)? Currently I don't see anything like dots or asterisks appear in the serial console, when the hsync frequency spikes or vsync is injected/missed. IDK if you're already listening for all relevant interrupts (meaning the chip doesn't generate an interrupt when it glitches out) or not (in which case I should check for what interrupt the chip self-reports).

Looking in the register definition, I find multiple SoG/csync width related registers by finding `separ`. I wonder if these registers are configured wrong, the TV5725 is seeing a spurious hsync pulse that shouldn't even be registered in the first place, my chip is damaged somehow, or the Wii itself is generating a bad signal.

In the meantime I may be researching playing games on Dolphin on LCD/VGA, or building/buying an alternative YPbPr-to-VGA transcoder. I hear that jamcoders (from wakabavideo) using the LMH1251MT-NOPB chip will rarely "drop sync on dark to light flashes", which is the sync bug everyone else is getting, unlike my random PLL screwup seemingly unaffected by image content or SoG capacitors.

ramapcsx2, 2023-06-05:
It can be anything or nothing you wonder about. That is the nature of this analog stuff.
You just have to best-guess what could be the cause and try to fix it methodically: One fix attempt / change at a time, then watch whether anything improved.

nyanpasu64, 2023-06-05:
I did find that if I increase the `SP_H_PULSE_IGNOR` register, I get frequent hsync frequency errors (steps), as well as the frequency jumping and sliding back to the original frequency (randomly or upon vsync). I'm guessing this is another way to make the chip see missing or duplicate hsync pulses, which, and the register isn't the direct cause of the bug. I haven't fully investigated the other registers yet; for example I want to look into `SP_L_DLT_REG`, `SP_DLT_REG`, and really the entirety of the sync register block.

> One possibility I considered was power delivery failure. I had difficulty probing the pads with a multimeter, and would get inconsistent voltage readings, but eventually I got a good reading of the GBS-C powered on with my usual USB charger, directly at the 5V barrel jack, and saw >4.9 volts. I'm guessing the issue I have isn't caused by a too-weak charger (though it could very well be caused by switching-mode noise or impulsive spikes making it to the sync processor).

I want to probe the GBS8200's power line with my PC audio jack, to look for noise. But I don't know how to connect it to the power line (either input 5V, or the reduced 3.3V or 1.8V lines), with a RC highpass filter to avoid exposing my audio jack to the full DC voltage. (I think my motherboard has its own series caps, but who knows.) I'd need some sort of tap PCB (like protoboard) connected somewhere on the GBS-C with wires or detachable connectors, going into AC coupling components, and hooked up to a RCA jack or cable to my computer. (The single RCA jacks I purchased from Amazon won't fit in protoboard holes, but I managed to crudely chop the pins lengthwise so they barely fit.)

Though now I'm dealing with a boot drive failure of my main desktop... on the day my PCIe capture card is set to arrive. So I have more pressing matters to deal with than debugging the GBS-C.

ramapcsx2, 2023-06-06:
Heh, oki, that is the right approach now :)
Forget about the power thing. Power is going to be suitable, not likely to be an issue.
The sync registers are though. They are important filters and detector settings, sync slicer and comparator stuff, that has to fit the source signal.
As these are barely documented (usually just a name and a text blurp without context), it is difficult to configure..

nyanpasu64, 2023-06-07:
> now I'm dealing with a boot drive failure of my main desktop...

fingers crossed it was a sleep-wake issue. I tried doing a BIOS update... which erased my boot order and overwrote my systemd-boot dual-boot menu with Windows Boot Manager. I had to fetch an Arch live USB to run efibootmgr to fix it... remember when they were called live CDs? does anyone use CDs anymore?

hoooooooooooooo boy

[image]

The top channel is a 28 MHz capture (using [CXADC](https://github.com/happycube/cxadc-linux3) of the Wii's output luma+sync signal fed to the GBS-C. The bottom 2 channels are a 192 kHz capture (using motherboard audio) of the GBS-C's output VGA blue line, as well as pseudo-csync.

My 28MHz capture revealed that before output hsync is lost, the _input_ component signal gets completely scrambled by a burst of high-amplitude noise in the 8 MHz range (possibly aliased from a higher frequency). This was completely missed in my 192khz audio capture.

- Unfortunately the 28 MHz capture card has capture latency varying by *multiple frames*. I had to align it with the low-latency audio capture, using the first/last active lines to align the signals modulo a frame.
	- Because the CXADC capture latency varies by multiple frames, I don't actually know if the input interference and output sync error occurred on the same frame or not. But I think it's the most logical interpretation of this signal, that the input interference caused the PLL to lose sync around 1 line later.
	- Is anyone here good with Linux PCIe drivers (including the [cxadc driver](https://github.com/happycube/cxadc-linux3/blob/master/cxadc.c) which creates a virtual device file), and knows how to make `pci_dev::irq` fire more frequently? Should I change the `pci_write_config_byte(pci_dev, PCI_LATENCY_TIMER, ctd->latency);` parameter to write a lower value, or will this result in the device getting short transfer durations and low bandwidth? For now I'm exploring various approaches to get data sooner, without waiting for the file read to be woken up by an interrupt.
		- I wonder if the current device-file API and non-blocking reads is enough to build a period-based (JACK-like) or delay-locked-loop-based (Pulse/PipeWire-like) capture system with predictable latency, similar to ALSA audio devices but at a higher sampling rate.

Any clue what's generating this loss of sync? My Wii? The GBS-C itself?

- If the GBS-C is generating noise, does it come from the TV5725, ESP8266, or power filtering?
- Does it do it randomly? Or in _response_ to loss of PLL sync? (I doubt this, because the noise burst happens _before_ the PLL loses sync, modulo a frame.)
- It could be the cxadc capture card as well, though I doubt it because I get identical symptoms without the capture card in place.
	- [ ] Just to be safe, I could also check that the bug occurs without my motherboard audio taps in place either, hooked up to luma *or* VGA output.

I currently have the GBS-C's VGA output hooked up to a VGA switch, then through an Extron distribution amp (active 2-way splitter) to my CRT monitor and my motherboard's line in.

- Could the VGA output chain induce high-amplitude noise, which becomes visible on input luma?
- How likely is it that noise in my monitor would make it through my power/grounding, EM radiation, or VGA cable, to show up on the luma signal? What about the Extron amp (though I also got sync loss without it)?
	- [ ] I'm thinking I should probe the blue/sync lines with my hi-res capture card, to test these theories.

ramapcsx2, 2023-06-08:
That 8MHz thing is surely a problem. I've no idea whether it actually comes from the Wii (maybe it can sometimes glitch?).
Of course the PLL will loose lock on that, too :)

nyanpasu64, 2023-06-12:
The analog power pins are shorted, either in the chip or on the PCB.

- [ ] Do we need more power supply bypass capacitors near the analog power pins? Or is the problem caused by a chip malfunction rather than power supply instability?

During GBS-C startup the 3.3V rail is *mostly* stable. In my testing so far, I did not encounter any video glitches, so was unable to measure the voltage rail during a glitch. After I capture 1-2 glitches, I may move my voltage probes to my next testing location.

----

How is the luma signal routed? So far my understanding of the luma circuit is:

- Luma has 75 ohms to ground/shield and 200 ohms to point `A`.
- Point `A` has C35 to !SOG0 (61) and C34 to G0 (62). I replaced C35 from a 100 nF capacitor to 1 nF, but the problem persists. I do not know the value of C34.
	- When the GBS-C is powered off, !SOG0 and G0 have <3.4 MOhm to ground (when positive) and 4 MOhm (when negative).
	- They have 10 MOhm to each other somehow.
	- I don't know what impedances they have when the GBS-C is powered on, or if these voltages are even actively clamped to ground during blanking intervals.
		- Perhaps the active voltage clamping functionality is malfunctioning and inducing voltage spikes and ringing?
- There may be other signal paths I missed, though I'm not aware of any.

Oddly this doesn't match the TV5725's docs, which do not call for resistors between the input pins and point `A`.

[image]

How should I test this system?

- [ ] I could measure across the 200 ohm resistor, to see if !SOG0 or G0 are being actively jerked around by the GBS-C rather than following the input signal.
- [ ] I could measure across C35 or C34 as well, or measure pins !SOG0 or G0 compared to ground (though in both cases, I'm worried the extra components attached past the AC-coupling capacitors will cause the circuit to artificially misbehave).

It may be worth also measuring chroma (either at the input, or across/past the 200 ohm resistor) to see if it's experiencing the same oscillations when sync is lost.

nyanpasu64, 2023-06-13:
can't measure across 200 ohms without a common-mode blocking choke/transformer. maybe someday i'll learn how transmission lines work. i'm tired.

can't measure which way the interference is propagating down the luma line (to test whether it's the Wii or tv5725 at fault) without a second high-bandwidth capture channel. don't know whetoer to buy a second pcie card plugged into a nvme slot, or a real oscilloscope with time-synced capture channels. or maybe just stop trying.

for now, i'll try buying a cooling fan for the tv5725. fingers crossed. (in hindsight i should not have installed the clockgen on the heatsink.)

ramapcsx2, 2023-06-13:
Active clamping in operation, so you can't measure anything regarding impedance then.
I would recommend you check out a second GBS board, second Wii, other pair of cables, etc :)

nyanpasu64, 2023-06-17:
video dropout occurred with active fan cooling

only thing left for me to do is to figure out if the interference is coming from the wii or the scaler, whether by subtracting measurements across the 200ohm resistor, or by measuring signal propagation down the cable

i could try dvd players in 480p component, but it's hard for me to sit through a movie

my main PC has started waking by itself in Windows. i suspected electromagnetic interference, but found out my computer woke twice without my wii losing signal once.

ramapcsx2, 2023-06-17:
You're loosing focus of the easy to test things, and I really recommend you get another GBS :p

nyanpasu64, 2023-06-18:
Huh... if it turns out the second GBS is also bad I don't want to invest even more money into defective equipment, if the first GBS was bad then investing money into equipment with a high failure rate (judging from the issue tracker) doesn't inspire confidence, and if the Wii is bad I don't want to invest more money into scaling a defective console (I might run Dolphin instead of buying another Wii?), unless I can find a way to block the interference. At least oscilloscopes will be useful in the future I hope?

is it possible I damaged my first gbs in installation? i don't recall cooking any chips... but idk anymore

nyanpasu64, 2023-06-19:
> I really recommend you get another GBS :p

ok ok i'll bite. I've been trying to get Dolphin running, but noticed that no matter what OS and graphical settings I pick, it seems to have more latency than my Wii. So I might need a scaler/transcoder anyway, for testing and/or playing games. The current GBS works for brief testing, though I no longer want to use it for actual gaming. TBH I'll probably delay ordering a new GBS8200 until my scope arrives and I confirm the TV5725 is producing the video interference.

- [ ] Is it at all a good idea to swap my existing ESP8266 and clockgen to the new board, to reduce costs and effort, or am I possibly bringing the defective components with me?
- [ ] I should probably check that sync loss still occurs with the clockgen disabled and ESP unplugged.

The previous GBS8200 I bought was from https://www.ebay.com/itm/284784624645. Is it possible that a different seller (eg. https://www.ebay.com/itm/111867300324) will have more or less issues? (Oddly *both* listings come from "Monroe Township, New Jersey, United States".) Or did I fry the board myself by running it in a hot enclosure, and it's unrelated to the place I'm ordering the board from?

It would be unfortunate if the new GBS8200 has the same issue, and I have no choice but to get a wakabavideo jamcoder.

nyanpasu64, 2023-06-21:
## debugging with a scope

(edited down from Discord chat logs)

(photo of scope on my desk) a bit late to ask, but should i have gotten a siglent sds1202x-e instead of rigol ds1054z? i hear it's less channels but better bandwidth and software

> most important thing is if it can record data for longer periods
> if it is and the bw doesnt bother you then meh
> 4 channels has its uses

- [ ] also perhaps i should unlock 100mhz on my scope
it'll help capture the signal, but i haven't fully researched the procedure and done the steps

https://cdn.discordapp.com/attachments/1083378131332763750/1120990591245889587/DS1Z_QuickPrint1.png
these normal oscillations have a period of 10 ns, so 100mhz
idk about the sync loss oscillations
this noise happens continuously, but only rarely does it reach a high amplitude to drown out the input signal

https://cdn.discordapp.com/attachments/1118186176801685584/1120999271865516112/DS1Z_QuickPrint4.png
- [ ] not the prettiest output hsync pulses either, not sure if it matters

caught a likely sync loss event (though i had my crt off so i don't know)
https://cdn.discordapp.com/attachments/1118186176801685584/1121013648350003270/DS1Z_QuickPrint22.png
seems to be a 100mhz noise burst followed by lower-frequency ringing which I caught on cxadc, which repeats.

(photo of luma line and gbs-c sync input, separated by 200ohm resistor)
it's odd to me the amplitude is just about identical on the two sides of the 200 ohm resistor, with the other side having 75 ohms to ground plus a 75ohm cable to the Wii
is the noise not coming from the video/sync pin? then why does it not occur when i unplug the gbs-c's power jack? 
does the gbs-c have something that isn't past the 200 ohm resistor? 

https://cdn.discordapp.com/attachments/1118186176801685584/1121015953883414528/DS1Z_QuickPrint24.png
(photo of luma and chroma lines) luma and chroma noise appears correlated
(the chroma is missing because this is vertical back porch after vsync)

I noticed something weird
i always get noise bursts on output vsync begin and end, but they're small and insubstantial
but when i plug my esp in, i also get random noise bursts all over the place
unplug the esp and they disappear

 

(sees micro-usb cable between ESP and computer, unplugs from ESP, and the oscilloscope screen suddenly freezes)
**AAAAAAAA don't tell me it's the extra long micro-usb cable i'm using to connect to the computer**

electromagnetic interference strikes again

i also get interference with the cable plugged into the ESP but not the computer. so the cable is acting as an antenna, not a ground loop.

## Thoughts

I'm guessing the long USB cable was acting as an antenna and picking up ambient noise.

- [ ] i'll close this issue once i spend a few hours playing a game with the USB cable disconnected, and get no signal loss (later, not now).

Why did I leave a USB cable connected to the ESP in the first place? In March I was hunting the flash corruption and reboot bug, and needed a cable connected to the GBS-C to get crash logs. Since a computer probably does not supply enough USB current to run the GBS-C, **I cut the 5V line on the ESP8266's PCB**, then found an extra-long micro-USB cable to run from the ESP to my computer several feet away (and coiled up part of the middle to reduce the extra length).

As a result, this cable is only connected to the GBS-C by ground (shield) and data lines, not the 5V line.

----

oddly, (IIRC with the USB cable disconnected) the scope still picked up two isolated voltage spikes on my scope from *rolling my chair across the floor*, and a longer ~90MHz voltage oscillation when I wasn't looking at the screen and don't remember what I was doing, followed by 2 voltage spikes from petting my cat. IDK what caused them and I can only hope they won't make my display cut out in-game.

Perhaps the GBS-C would benefit from a metal shielded case and ferrite beads on video/power cables?

- [ ] I could go looking for a metal cookie tin to put the GBS in, though I don't know how cables will get in and out, and I'm concerned about cooling.
- [ ] Also I have a package with ferrite cable clamps arriving in a few days/weeks.

## ESP power cable interference

[image]
[... 9 lines cut ...]
After unplugging most electronics, turning off my lights, unplugging my power strip with the flickering neon bulb (so I can only power my Wii and scope, not CRT), unplugging my electric toothbrush's induction charger, the noise persisted when my USB cable was plugged in.

I decided to plug a shorter USB cable coiled into a loop into the ESP, and aim it in various directions to find the source of interference. Sadly I did not find any direction with a decrease in noise.

I set the scope to probe the Cb channel and trigger off Priiloader's colored text on an otherwise B&W background. When I zoomed out to see several frames at a time, I noticed that the interference appeared stable but drifting leftwards (shorter period than Wii) at around 2.5-3 seconds per cycle, meaning the interference is periodic at around 59.94 + 0.4 Hz (unless one of my assumptions was inaccurate for some reason).

I noticed the periodicity only appears when leaving the GBS on for 30 seconds or so, and the interference is much larger in amplitude/intensity while the GBS-C is powering on (but practically disappears if I unplug the USB cable). **This would indicate that the GBS-C itself is generating interference** (in addition to EMI from house wiring and such), **b [... cut ...] strong on both sides of the 200 ohm resistor) to oscillate at around 8 MHz afterwards.

- [ ] Hopefully using a shorter USB cable with a ferrite bead will prevent these issues.

And some parting thoughts:

- fuck electromagnetic interference.
- i still don't know how electromagnetism works in circuits, and I feel I can trust nothing (in a sufficiently big/fast circuit, not even different points connected with a conductor are at the same voltage).
- turns out i did need an oscilloscope after all.

nyanpasu64, 2023-06-23:
> Unfortunately I cannot save images onto the flash drive since the 1054Z does not recognize either of two flash drives, both a 32GB one that used to work and a smaller 4GB one (people say smaller flash drives are better), both clean in fsck. If I tried to reboot the scope I'd lose the saved traces of this incident.

- managed to get the signal off using ultrasigma/scope on windows, export to csv, then reboot to linux and open the csv in pulseview (windows stable pulseview opens a dialog and hangs, and windows nightly pulseview is an errored file). since then i've rebooted and the 4GB flash drive works.
	- i still have issues saving a waveform to csv on a flash drive, unless i first stop capture then unplug/replug the flash drive.

IIRC I had taken that recording (`wii sync drop.sr`, not uploaded) when I fed my Wii's signal through a luma extension cord, with the intent of recording both sides of the luma cable, but I had not done so. That explains the missing CH2 in `Waveform.csv` (the source file), which was (at the time) recording the Wii end of the luma signal, but I'm not confident. I had probably coiled the luma cable extension so it took up less space, and the extension could end (at the GBS-C alongside chroma cables) near where it started (the luma cable which can't be separated more than a few inches from the chroma cables).

## sync loss recording

since then i've taken a 250 Msps (4ns interval) capture of the wii's sync drops, and imported it into PulseView: [wii sync drop 2.zip](https://github.com/ramapcsx2/gbs-control/files/11841916/wii.sync.drop.2.zip). the channels are:

1. wii luma out
2. gbs luma in
3. gbs luma ground - wii luma ground
4. trigger

based on this trace, i'm guessing about the cause of the sync loss... the wii does not take kindly to electromagnetic interference being backfed into the video outputs, and will sometimes emit 8mhz oscillations. How do I know the oscillations come from the Wii?

- I was feeding the luma signal through a 96-inch RCA extension cable, and probing the signal at both ends of the cable (closer to the Wii and GBS-C). The video signal appears around 3 samples (12 ns) earlier at the Wii end than the GBS-C end.
	- To calculate the theoretical delay, I found [a site](https://web.physics.ucsb.edu/~lecturedemonstrations/Composer/Pages/76.18.html) which says "In coaxial cable, the speed of an electrical signal is about 2/3 of this, or about 8 in/ns." 96 inches / (8 inch/ns) = 12 ns of delay across the cable. This agrees with my scope measurements.
- To find part of the signal with only noise, I can check during the hsync or back porch areas of the 480p luma signal, which should be perfectly flat aside from interference.
	- The 8MHz oscillations also appeared to be 2-4 samples later at the GBS-C than the Wii, indicating the Wii itself was outputting them. Though this is a best guess because the noise has a lot of noise or aliasing, and doesn't align perfectly between the two signals.

- To check for skew between the oscilloscope (DS1054Z) channels, I tried probing a high-resolution VGA signal at 500 Msps using two probes at the same time. I found that the two channels were just about perfectly aligned, but wiggling the *ground clips* can cause phase shifts up to ±1ns or so.
	- [ ] One workaround is to use a sharp probe and a high-frequency ground spring, but AFAIK those don't stay in place and you have to hold onto the probe, so I can't use it when I need my hands elsewhere? IDK.

On the other hand, there's initial higher-frequency (and probably quite aliased) noise before the 8mhz oscillations. 250 Msps is just not enough to capture it, and I haven't yet ran a capture at 500 Msps with just 2 channels.

- Capturing at 500 Msps requires only enabling 2 channels, since any more brings you down to 250 Msps. Unfortunately, even a third channel used only for triggering and not recording seems to count too, meaning I can't trigger off my Pi Pico programmed as a hsync/vsync error detector, and must trigger off video instead. Since the Wii's luma output is AC-coupled and I can only set a fixed trigger level in my scope, if I wanted to reliably trigger off noise but not regular video, I'd have to show a synthetic test image with consistent average DC offset (eg. 240p Test Suite showing a black/white screen), and power off my CRT monitor so I don't needlessly wear it out. This is very doable, but I haven't gotten around to it yet.

My theory is that the electromagnetic interference arises because my luma extension cable was coiled into 4 loops using over-under coiling, which minimizes cable strain but unfortunately acts as a loop antenna. And this cable was picking up noise (from an unknown source) and sending it as a common-mode signal between the Wii and GBS-C.

- My third oscilloscope probe measured the voltage between different points on the luma cable's ground/shield (at the 2 ends of the extension cable). In a world of spherical cows, this voltage would always measure 0 volts because the entire cable's shield is connected by a conductor. In practice, every time the luma signal oscillated, it was preceded/accompanied by up to 1 volt (possibly more but clipped) of peak-to-peak of interference *between points on a ground conductor*.
	- Unlike the 8MHz ringing, this initial noise spike's high frequencies was *not* delayed in phase between the Wii and GBS points of the cable (though the low-frequency 8MHz components seem to be later at the GBS). To me that sugg [... cut ...]
	- i'd have to recreate the coiled cable to increase my odds?
- [ ] Hook up coiled luma cable to DVD player, and watch to see if high-amplitude noise appears and/or sync is lost
	- Do I measure ground voltage difference, my sync loss trigger signal, both amplitudes, or all 4 (at a lower sampling rate)?
- [ ] remove coiled luma cable and see if Wii still loses sync
- [ ] find out where in my room this accursed noise is coming from... heat-damaged gbs? audio mixer? something else?
- [ ] add ferrite clamp on Wii's luma cable, or all signals (not sure)

nyanpasu64, 2023-06-24:
the issue is not caused by the gbs. The wii still outputs noise bursts when connected to an unpowered gbs, or even to a dumb 76 ohm triple RCA terminator I made with the shields connected together (I didn't have 75 ohm resistors). I will continue to investigate to see where it's coming from, see if the problems occurs with my custom audio circuitry unplugged, or USB hard drive and GC controller, etc.

closed 2023-06-24 by nyanpasu64

ramapcsx2, 2023-06-24:
Man.... this'll be one epic tale once you're through xD

nyanpasu64, 2023-06-25:
i've traced down the video noise to my wii itself

i get the noise from the wii's video output alone, with no controllers or hard drives plugged in, and just a SD card and a video output (no audio connection) leading to a triple 75ohm terminator and the scope probe. all but one scope probe was removed (so no extraneous ground paths, though the scope is grounded and my computer/monitors have many ground loops). guess it's time to move on to different hardware for playing games (as opposed to messing with Wii channels casually).

I was surprised the Wii was malfunctioning, because I've been using this (and another) Wii for years on various other LCD displays (and even my CRT TV for some time), without noticing any video dropouts, and nobody I've talked to has reported seeing Wiis with oscillation or sync loss on the video output. Perhaps my hardware is aging (or has overheated from RiiConnect24 standby) and started to degrade, or perhaps other displays' sync separators are more resilient to absurd voltage spike oscillations and sync loss on the SoG/luma signal.

## Wii alternatives 

unfortunately for playing GC/Wii games on PC, dolphin has inconsistent extra input latency with vsync on, because it schedules emulation timings based on wall-clock time (periodically sleeping the emulation thread to loosely match Wii timings) rather than monitor vsync. By probing in Tracy Profiler, I found that if you enable vsync, then the emulator will sometimes finish rendering frames too early relative to GPU scanout, then wait up to 12+ milliseconds for room on the swap chain (IDK how many frames long) to display it, then speedrun emulating the next frame (too early relative to GPU scanout) because it fell behind on wall-clock time while waiting for room on the swapchain. The Dolphin devs recommend VRR to minimize latency without tearing, but the real Wii doesn't need it for sub-Dolphin latency, and my CRT monitor can't do VRR without the image shifting/squeezing vertically.

- i may be able to tweak the wall-clock scheduling to synchronize on scanout, but i assume it will be an uphill battle to get upstream to accept this change.
- additionally I don't know how Dolphin queues/resamples audio, between the emulator generating sound to the system playing it. I know different audio backends have different latencies (Windows WASAPI exclusive is extremely low latency, everything else ranges from moderate to terrible). I want to research how it works, even if it's not directly related to video scheduling.
	- Does Dolphin in exclusive WASAPI mode keep audio buffers smaller than a video frame (relying on how the game emulator generates audio at the same wall-clock rate the audio interface consumes it), and resample as needed (possibly using a PipeWire-style delay-locked loop)?
	- I implemented the GBS-C's external clockgen frametime lock as a first-order delay-locked loop without thorough knowledge of control theory, but it seems to work adequately (if a bit slow to converge) if it detects the input/output rates accurately to begin with. PipeWire's audio scheduler (and presumably resampler) uses a second-order filter which seems to overshoot if exposed to a large initial error, and I only understand its behavior empirically.

I've also been suggested to buy a Switch lmfao

ramapcsx2, 2023-06-26:
I'd still try more Wiis and see whether they all produce this. Also it might matter whether homebrew is used, which might mess with the fully customizable DAC chip it uses for video.

nyanpasu64, 2023-06-26:
For the longest time this room has had a 3-way dimmer switch for its ceiling lights, which kept giving me trouble. When I turned the slider up, there was a point it would suddenly dim the lights and take a good chunk out of the LED bulbs' light output waveform (relative to 120hz rectified AC power). Additionally the ceiling lights would randomly flicker when set too bright.

2 days ago I finally got it replaced with a non-dimmer switch, which seems to have solved the light flickering. Inspecting the dimmer switch, I discovered it was an "incandescent dimmer" now hooked up to 120V LED bulbs, with burnt flux residue on the solder joints (whether from soldering or from overheating in use), blackened plastic around a giant metal coil (indicating overheating?), smaller capacitors under the PCB, and was hooked up to LED bulbs. In hindsight hooking an active incandescent dimmer to active LED bulbs with rectification and whatnot, and using it while it was this burnt, cannot have possibly been a good idea.

[image]

I also found a neon bulb which lights the switch when it's turned off (and possibly on? IDK.) Of course I generally run my CRT with the room lights off to avoid glare... The bulb electrodes look somewhat corroded, and it's possible either the giant coil, the neon bulb, or both were sporadically radiating EMI.

[image]

In the testing I've done since removing the bad wall switch, I have not seen one incident of noise and sync loss. Though yesterday I accidentally ran testing with my scope set to AC coupling on the "sync loss" signal, meaning I was unable to confirm my sync loss detector was actually outputting low and not high all the time (by force-triggering to view the current voltage level). And when I shut down my scope for the night, I dislodged the VGA connector, meaning it's possible the sync loss detector signal was high the whole time and I never noticed.

As a result I've been doing more testing today. Again I have not yet seen a sync loss in gameplay or using the scope to watch for the sync loss signal, although I am still running the scope for a few more hours to capture sync loss events if they recur. I'd cautiously call this issue fixed?

Though I do get "normal" sync loss when switching from Priiloader -> Homebrew Channel to System Menu for the first time, or when booting Super Mario Sunburn from Nintendont (even with Force 480p on, the "enable progressive scan?" dialog does not output a valid 480p signal, unlike some other games I've tested).

## Conclusion

I currently believe that a defective room lighting dimmer switch was emitting sporadic bursts of loud electromagentic interference (most likely EM waves), either from the coil, the neon bulb, or wall/ceiling wiring. The lighting and Wii power are on separate circuit breakers, but I've heard it's possible the wires are routed in parallel and interference gets coupled between the two circuits. This electromagnetic radiation then interfered with the Wii and/or long cables in the room, causing the Wii's more sensitive circuitry to emit 8MHz bursts of interference out the luma cable, causing the GBS-C to lose sync. It may also have led to my desktop computer spontaneously waking up. Presumably the DVD player's component output was less sensitive to interference and didn't produce the same signal errors.

This would explain why the Wii never had problems prior to moving into this room. It does not explain why it worked fine on my CRT TV, or on the GBS-C before I switched to bypass mode (possibly those devices are less sensitive/obvious to sync loss, or perhaps my light switch was deteriorating and only recently started emitting radiation despite already flickering?). I still do not know why replugging the GBS-C caused the luma cable to have lower-pitched bursts of interference, and whether they were coming from the Wii or GBS-C.