ramapcsx2/gbs-control #462: scanlines with 480i

filevans, 2023-05-13:
hello, please could scanlines on 480i be added? like the ossc and retrotink 5x can do
thank you

YoImLee, 2023-05-13:
If you're talking about alternating scanlines to get a 480i crt look, gbs control can already do this.

Set the deinterlace method to bob, turn on the line filter and scanlines and you should get a similar effect.

filevans, 2023-05-13:
no,   i want 480i to look like 240p with no interlacing or effects, just scanlines, like the ossc can do

filevans, 2023-05-14:
this it how scanlines look on the ossc in 480p:

[image]

nyanpasu64, 2023-05-14:
You can enable scanlines on 480i content by enabling scanlines and picking bob deinterlacing. Note that with the current emulated scanline filter, chroma is not properly reduced on dark lines, in 240p and probably 480i as well (#458).

If you want scanlines on 480p content... go buy a VGA CRT monitor I guess

filevans, 2023-05-15:
> You can enable scanlines on 480i content by enabling scanlines and picking bob deinterlacing. Note that with the current emulated scanline filter, chroma is not properly reduced on dark lines, in [...]

If you bother to read the messages above properly before posting, all that has already been said, look at the second message. As already stated there are no proper scanlines visible on 480i with Bob method, they are very weak scanlines on an interlaced picture

nyanpasu64, 2023-05-15:
480i video games (or broadcast television) did not have visible scan lines on real CRT TVs, unless you were moving your eyes up or down. Standard-definition televisions were designed so each field had scanline gaps (except on tiny blurry TVs), but each field's scanlines filled the gaps of the previous field, producing a seamless image for non-moving objects.

Are you trying to add emulated 240p scanlines on 480i and 480p content, by discarding or blending away every other line and replacing it in the output with a black line? That's a different requirement from simulating how a 480i TV scans. If you want this, you could try clicking the 15KHz button on the web UI to generate a 240p output signal, but a LCD monitor may not support it, and will not add scanlines. Alternatively you could use passthrough, 480p, or 960p scaling, and hook up the GBS-C's output to a VGA scanline generator you buy elsewhere.

Can you force-reconfigure your video source to output 240p rather than 480i?

filevans, 2023-05-15:
No I just was hoping to get scanlines on 480i from PS2 and Xbox like the OSSC can do. If you see the comment above I said that if you hold the RGBS sync button halfway in and halfway out it puts scanlines on 480i content, albeit on a frozen picture, I thought maybe something could be exploited there, just an idea

filevans, 2023-05-16:
No I just was hoping to get scanlines on 480i from PS2 and Xbox like the OSSC can do

if you press the RGBS sync button in then press it off / our again but hold the button halfway out it puts scanlines on 480i, I thought maybe something could be exploited there, just an idea, look:

[image]

This is scanlines on 480i albeit on a frozen picture

invaderlex, 2023-05-31:
You are better buying one of this and use 960p:

[image]

filevans, 2023-06-01:
> You are better buying one of this and use 960p:
> 
> [image]

"these" is that their official name? don't think i will be able to find it on google with just that

4lexei, 2023-06-04:
Hey, people are trying to help. Taking into account Nyanpasu64's continuous contributions to this project, I think he has the right to answer even if his solution has already been discussed before.

That's a scanline generator, maybe based on the slg3000 (not sure). A very common device easy to build and also probably available on Aliexpress.

filevans, 2023-06-11:
> Hey, people are trying to help. Taking into account Nyanpasu64's continuous contributions to this project, I think he has the right to answer even if his solution has already been discussed before. [...]

a name of a device to search for is what people need, not just "these"

ramapcsx2, 2023-06-11:
Enough good information has been given to you for free here, @filevans.
This topic is done.

closed 2023-06-11 by ramapcsx2

filevans, 2023-06-11:
@ramapcsx2 i also provided good information too, look at my screenshots, ideas and feedback

invaderlex, 2023-06-15:
> @ramapcsx2 i also provided good information too, look at my screenshots, ideas and feedback
> 
> also i could never find the device he showed a picture of that output 960p, as he didn't say the name of it or where to get it, i could only find one that has: Resolution: 1024x768, 800x600

Hi, that is a generic scanline generator, that is its name, you can find them on ebay, aliexpress typing SLG or scanline generator, is based on the slg3000, you put it in the output of the GBS and it will make half of the lines darker, I recommended using 960p because it is an integer scale 480x2 and one of the switches is for double scanline lines so it will make 2 lines dark then two lines untouched giving the scan line look at higher res. If you use 720p or 1080p the dark lines will be too thin and not in the correct place over the pixels. You can build you own is an easy project, the theory is that one of the IC function as a sync time divider, it takes the sync signal and divide the sync pulse by 2. the other IC function like a solid state switch when one pulse is detected it will send some of the signal to ground making the line darker, and because the pulse is divided by two it will alternate between one dark and one line passing like normal.

invaderlex, 2023-06-16:
Mine

[image]

filevans, 2023-07-11:
> Hi, that is a generic scanline generator, that is its name, you can find them on ebay, aliexpress typing SLG or scanline generator, is based on the slg3000, you put it in the output of the GBS and it will make half of the lines darker, I recommended using 960p because it is an integer scale 480x2

The scanline generator I found on AliExpress only outputs Resolutions: 1024x768, 800x600. It says. So do you mean use 960p on the GBS-C? As you don't explain properly. The generic scanline generator doesn't seem to output 480p to be doubled to 960p either

invaderlex, 2023-07-18:
> > Hi, that is a generic scanline generator, that is its name, you can find them on ebay, aliexpress typing SLG or scanline generator, is based on the slg3000, you put it in the output of the GBS and it will make half of the lines darker, I recommended using 960p because it is an integer scale 480x2
> 
> The scanline generator I found on AliExpress only outputs Resolutions: 1024x768, 800x600. It says. So do you mean use 960p on the GBS-C? As you don't explain properly. The generic scanline generator doesn't seem to output 480p to be doubled to 960p either. What resolution do I set the generic scanline generator to and what resolution the GBS-C

Look for the one that looks like mine in the pictures. I see that the description of the product on ali says 1024x768, 800x600 but it works with any resolution as it is only a fancy Solid state switch sending half of the lines to ground. Set the GBS-C to output 960p and the SGL for double line mode (WIDTH Switch).