ramapcsx2/gbs-control #463: 1920 X 1080 too narrow

filevans, 2023-05-13:
Hello, when I choose 1920 X 1080 resolution the display is very narrow on the screen, even though the console is 16:9, it is displaying on the monitor 
even narrower than 4:3, is there anyway to fix that? 

The native resolution of my monitor is 1920 x 1200

I tried to stretch the image in the position and size settings but you get garbled green stuff come when you widen it
[image]

Thank you

YoImLee, 2023-05-13:
As far as I know 1920x1080 cannot be stretched and is really only meant for 4:3 sources on 16:9 displays. If you want to use a 16:9 source your best bet is probably 1280x720.

filevans, 2023-05-13:
It wasn't just on 16:9 though, the 4:3 source was too narrow as well and couldn't be stretched. On 1280 X 720 it didn't fit the height so good, there are big borders top and bottom and the scanlines were uneven when trying to stretch the vertical. Whereas 1920 X 1080 produced even scanlines when I stretched the vertical

YoImLee, 2023-05-13:
Perhaps your 16:10 monitor has an option to preserve the original 16:9 or 4:3 aspect ratio? You could also consider trying the line filter to mitigate the unevenness when stretching the image.

filevans, 2023-05-13:
Yes the monitor has those options but when using this resolution and certain others it doesn't allow those options to be selected for some reason. I'm using DVI, I don't know if using VGA instead might help?

filevans, 2023-05-14:
> Perhaps your 16:10 monitor has an option to preserve the original 16:9 or 4:3 aspect ratio? You could also consider trying the line filter to mitigate the unevenness when stretching the image.

another thing, when i try selecting 1280 X 1024 it always dismisses it and moves to 1280 X 720 instead, and I would like to try that resolution out, is there any way to fix that? thanks