ramapcsx2/gbs-control #464: Can't connect through USB

gekkedaan, 2023-05-23:
I am unable to update the firmware because my computer doesn't recognize the GBS-C. I connected it through a micro-usb cable, powered the device on, but it simply won't get recognized. I tried different USB slots to no avail. The device seems to light up when I connect the USB to my PC, but the PC itself won't react to anything. Can someone help me along? I just bought this device brand new and immediately I run into this issue.

4lexei, 2023-05-23:
Hello! Connect your device via USB and try to install this driver: https://4lexei.com/CH34x_Install_Windows_v3_4.zip

richms, 2023-05-23:
Try another cable, because you probably have a faulty or data only cable.

nyanpasu64, 2023-05-25:
Are you on Windows? Try opening up Device Manager and post a screenshot here.