ramapcsx2/gbs-control #466: "Signal out of Range" on AVerMedia capture card (Live Gamer Extreme 2)

einso, 2023-05-27:
Hi,
I am having a lot of "Signal out of Range"/"No Signal" errors showing in my capture cards interface using my GBS-C. This seems to primarily come into effect with PAL60, but not exclusively (for example, Ridge Racer Hi-Spec demo displays as NTSC 60hz on my end, even when running on a PAL system). The capture card supports 2160p, 1080p, 1080i, 720p, 576p, 480p, 480i, and I am running at 1920x1080, so this shouldn't be an issue.

Is this a known issue and I am simply doing something incorrect, or is this some incompatibility caused by my setup?

4lexei, 2023-05-28:
Capture cards are really picky with retroscalers What are the output frequency of the GBS? My old Avermedia Live Gamer Mini is not compatible with frequencies below 59.80hz, and almost all my PAL consoles modded to 60hz were below that. If I remember correctly: SNES 59.55, Saturn 59.41… Now I have Elgato’s HD60 X, amazing device, everything works without adjusting the HTotal.

VGA to HDMI adapters gave me some problems in the past too. Some of them are not 50hz compatible.

By the way, If I remember correctly, NTSC games will run at 60hz in PAL systems. Is your RR Hi Spec disc NTSC? That would make sense, PlayStation works this way: your game dictates the region/hz, not your console.

einso, 2023-05-28:
Did some testing, it doesn't seem to be related to the game - PS2 menu sometimes work, sometimes throws the same "out of range" error, even after resetting everything to default in the web GUI. Video timings, according to web GUI, are 49.999XX for source and out (PAL system, so expected), resolution still 1920x1080. It seems to be an issue with recording it primarily, as the passthrough on the capture card works just fine. Tested this with OBS and the AVerMedia software.

Territorialist, 2024-02-28:
I'm having the same issue here, I get no signal. Does anyone have a solution or alternative?

Territorialist, 2024-03-03:
> I never found a solution, I just ended up getting an Elgato capture card instead. That one works fine. Sorry to say, because that isn't really a solution.

That's a shame, might have to do the same. What was the model of Elgato Capture Card you bought?