ramapcsx2/gbs-control #467: Some consoles are very unstable

alucryd, 2023-05-31:
Hi all,

I have a very annoying issue with my GBS control where some specific screens make the display completely crazy, but only on some consoles. It happens a lot on the NES (with Krikkzz' RGB Blaster) and sometimes on the Mega Drive, while a 1-chip SFC and an RGB modded French N64 are pretty much perfect. Thought it might be a PSU issue (all originals) but I since acquired a Triad PSU and the issue is still there. I'm using HD Retrovision cables exclusively.

Easiest way to reproduce is to load the 240p test suite on the NES, the main menu is almost unusable. On the Mega Drive, some of the 240p test suite patterns trigger the issue, and things like the Sega screen in Sonic do as well.

I'm seeing the following in the console when it happens:

```
FrameSyncManager::runFrequency() measured inconsistent FPS 60.098457 and 58.973782, retrying...
FrameSyncManager::runFrequency() measured inconsistent FPS 60.678001 and 60.098141, retrying...
FrameSyncManager::runFrequency() failed!
FrameSyncManager::runFrequency() measured inconsistent FPS 60.098366 and 60.678047, retrying...
FrameSyncManager::runFrequency() measured inconsistent FPS 60.098953 and 60.676987, retrying...
FrameSyncManager::runFrequency() failed!
FrameSyncManager::runFrequency() measured inconsistent FPS 59.530540 and 60.099224, retrying...
FrameSyncManager::runFrequency() measured inconsistent FPS 61.267986 and 60.098366, retrying...
FrameSyncManager::runFrequency() failed!
FrameSyncManager::runFrequency() measured inconsistent FPS 60.098049 and 61.268364, retrying...
FrameSyncManager::runFrequency() measured inconsistent FPS 59.530766 and 60.098186, retrying...
FrameSyncManager::runFrequency() failed!
FrameSyncManager::runFrequency() measured inconsistent FPS 60.098320 and 59.531116, retrying...
FrameSyncManager::runFrequency() measured inconsistent FPS 60.099270 and 59.530319, retrying...
FrameSyncManager::runFrequency() failed!
FrameSyncManager::runFrequency() measured inconsistent FPS 60.677082 and 60.099087, retrying...
FrameSyncManager::runFrequency() measured inconsistent FPS 58.973610 and 60.098545, retrying...
FrameSyncManager::runFrequency() failed!
FrameSyncManager::runFrequency() measured inconsistent FPS 60.098003 and 58.973652, retrying...
FrameSyncManager::runFrequency() measured inconsistent FPS 60.677723 and 60.098545, retrying...
FrameSyncManager::runFrequency() failed!
```

Any idea what the issue could be, or how to diagnose it?

rwebster85, 2023-06-16:
I've seen similar with the Mega Drive and disabling the external clock (requires reboot to take effect) and disabling Frame lock creates a stable image that seems to run perfectly.

alucryd, 2023-09-15:
Thanks, will try that soon. Also planning on disassembling my prebuilt aliexpress unit and reassemble it the official way, there are some funky things going on due to the inclusion of a composite input on mine. I now have RGB component cables for all my consoles so I don't need composite anymore.