ramapcsx2/gbs-control #469: Audio output volume?

therealteamplayer, 2023-06-08:
I'm using a pre-built GBS-Control unit connected to component video and RCA audio, and specifically the audio coming out of the HDMI port is too loud (compared to just plugging the RCA into my TV) to the point of occasional distortion. Is there a way to adjust the gain applied to the audio or am I better off connecting the audio to external speakers etc?

israpps, 2023-06-08:
GBS Control does not have audio related hardware.

So its not something we can help u. at least not without knowing the involved hardware?

invaderlex, 2023-07-18:
> I'm using the ODV model that's available on Amazon, I assume that the audio is stuck with however that system handles it? In a few weeks I'm moving somewhere where I have access to better external speakers so I'd be able to plug the RCA jacks directly into there, so it's not that big of a deal for me

You can always add a passive volume control in the RCA

[image]

You can measure the pot values and use a fixed voltage divider with 4 resistors (2 per ch)

nyanpasu64, 2023-07-20:
Which chip does the ODV use to convert analog audio to HDMI audio? I don't see any mention of HDMI audio or volume in the firmware download at http://www.pstbiz.com/odv-gbsc/software/gbs-control%208.17.rar.

If you can't fix audio volume in software/firmware you may need to use a voltage divider. Though before doing so, what's the load resistance of your custom PCB's audio inputs? That will affect how the potentiometer voltage divider will attenuate your signal.