ramapcsx2/gbs-control #470: Pre-built unit: no screen tearing on startup, but screen tearing after changing between 480i and 480p

therealteamplayer, 2023-06-18:
I am using a pre-built GBS-C unit (the ODV GBS-C) and have come across an issue wherein screen tearing is introduced only after switching resolutions from 480p to 480i/240p or vice versa. After switching, the console displays a 'no stored display clock' message that it does not display on a cold startup, and screen tearing that would otherwise not appear is introduced.

The makers of the ODV unit claim that it does include an upgraded clock, but my theory is that it is being used only after a cold boot, and no longer being detected after switching resolutions. Unsure if it is a hardware or software issue (I flashed the latest firmware from this GitHub page). I don't know whether this would occur in a unit that I would build myself, but as I don't own a soldering iron nor no how to use one I can't test that.

nyanpasu64, 2023-06-21:
- enable developer mode. in the console to the right, what message do you see after turning on frametime lock with clockgen enabled?
- does enabling frametime lock fix tearing? a few months ago i implemented a change to add functioning frametime lock with clockgen on, but I don't know if the ODV firmware predates this change (in which case frametime lock doesn't work with clockgen on).

therealteamplayer, 2023-06-21:
I just get!

```user command 5 at settings source 2, custom slot 65, status 5
FTL on
Active FrameTime Lock enabled, disable if display unstable or stays blank! Method: 0 (vtotal + VSST)
```
I'll note also that my TV has trouble with the frametime lock setting, wherein the display will black out every so often as if the output has just been changed.

therealteamplayer, 2023-06-21:
One other thing: I had managed in the past to get frametime lock to work correctly without dropouts, but this would only happen if I manually set HTotal such that the output refresh rate was equal to the input rate. This couldn't always be done though since I could change the rate only in set intervals: e.g. for 480p 1280x1024 I could set it exact whereas for 480p 1920x1080 it would be 0.007Hz off and cause issues.

nyanpasu64, 2023-06-21:
- [ ] Did you verify that "Disable External Clock Generator" is unchecked, and if you toggled it you rebooted before doing more testing? Then you enabled frame-time lock and got a message about "Method: 0 (vtotal + VSST)"?

If so, either they lied and didn't actually install a clockgen, or they're using a custom "clockgen" they didn't upstream and frametime lock doesn't recognize.

But in any case, the GBS-C *should* redetect and lock onto the correct refresh rate when switching input modes. I don't know why that isn't working in your case.

- [ ] If you unplug and plug the GBS-C, does it stop tearing?

> I'll note also that my TV has trouble with the frametime lock setting, wherein the display will black out every so often as if the output has just been changed.

- [ ] Did you try both frametime lock modes without clockgen? Did you try frametime lock with clockgen?

therealteamplayer, 2023-06-23:
yeah the disable clockgen is unchecked, if I remember correctly when I disabled and rebooted I got the tearing immediately rather than once I switched resolutions. frametime lock has always shown the same message when I enabled it.

Once tearing starts, I can stop it by turning the unit off and on again.

yes I tried both frametime lock modes.

therealteamplayer, 2023-07-06:
To update: I did some more testing and I can confirm that I get mild tearing every now and then regardless of whether I switch resolutions or not. It's like a tear line that slowly moves down the screen. So it seems like my device doesn't have the clockgen or it is not recognised.

Adamanimus, 2024-11-10:
I also got ODV-GBS-C recently. AliExpress sellers claimed it has clockgen mod but i don't see any indication of clockgen being active.
Every so often i get single acrean tearing line going down slowly and disappearing.
No persistent screen tearings. 
I can't find any DIY without clockgen mod. How are screen tearing on these without clockgen mod? Is there horrible screen tearings everywhere or just single line going down slowly occasionally?

Lot of AliExpress sellers are claiming ODV-GBS-C has clock generators and even shows board with arrow pointing to clock generators. Is it possible for it to work without being recognized in the firmware?

Totix92, 2025-10-21:
> Clock gen is broken on the ODV GBS-C. I2C SDA/SCL lines to clock gen chip were swapped.

Is there a way to fix this? Is it via software or hardware? If so, can someone tell me how to do it correctly to avoid damaging the device?

trejan, 2025-10-25:
>I have swapped the traces and now it works. 

Yep. You've done it. Looks good.

> But also looking at the traces where I swapped it, it seems to be going into the chip with larger air cooler on top of it. Does the trace go into that chip and do anything?

It is the I2C bus that the ESP8266 controller uses to communicate with Tvia video processor chip, OLED display and the clock gen. Cutting it where you did will fix the factory mistake where they swapped the SDA/SCL lines for the clock gen. 

> Would it cause any problems with there I swapped the traces?

No. It is only fixing the clock gen I2C lines. Everything else is already correct and unaffected as they're connected before the swap

> Also, gbs-c web control interface sometimes shows this messages. What does these messages mean? Is there anything wrong with it?

It thinks the frame rate is abnormal. It is usually caused by the console resetting or changing resolution.

If the log shows "Active FrameTime Lock enabled, adjusting external clock gen frequency", the OLED display is showing the menu and there is valid video output then it is fully working. If the I2C bus was shorted then nothing would work except for the web UI.

Adamanimus, 2025-10-26:
Ah so I was correct to cut and swap there. I thought that if SDA/SCL lines were swapped then maybe I should have cut closer to ESP8266 than cutting close to clock gen chip. That is assuring. Thanks.