ramapcsx2/gbs-control #473: remove "harmless" framesync frequency warnings, when debugging is turned off?

nyanpasu64, 2023-07-02:
- [ ] `Error: trying to tune external clock frequency while clock frequency uninitialized!`
- [ ] idk what else

ramapcsx2, 2023-07-02:
The warnings represent stuff I could think of or saw happening when developing stuff at the time.
Since a lot of code changed, and if those warnings don't show, it makes sense to remove and replace with anything new :p

nyanpasu64, 2023-07-02:
The problem is that "Error: trying to tune external clock frequency while clock frequency uninitialized!" *does* show even in normal operation, because the existing frame sync (and my clockgen frame sync) begins running before `externalClockGenSyncInOutRate()` synchronizes the external clockgen. So this warning may falsely alarm viewers, and perhaps should be disabled by default. Of course ideally you'd run `externalClockGenSyncInOutRate()` before `runFrequency()`, but I don't want to change the order of execution of code where I don't understand the current system for when tasks run.

If `externalClockGenSyncInOutRate()` fails (due to faulty hardware etc.), then this warning will loop infinitely. Maybe this is an annoyance for end users, and removing these prints won't hurt debugging(?) since `externalClockGenSyncInOutRate()` itself will print an error message. But I'm not entirely sure that's a good idea.

Perhaps the error should be skipped on its first print of a "session", or suppressed until `externalClockGenSyncInOutRate()` fails, etc. I don't know how long a "session" is, and when to count it as the "first print" again. I think I could've found out a month or two ago, but I'm again quite rusty on the codebase and state machines after a long period of not working on the software side of things. In the meantime I *could* disable the print, which should be a less-bad user experience than before.

ramapcsx2, 2023-07-03:
Hm, sounds like it'd be best to actually go in and make sure the order of operations is so that the frame sync skips until the clock gen has initialized.