ramapcsx2/gbs-control #475: My clock gen doesn't validate according to the doc

brandon-arnold, 2023-07-04:
Is the clock gen test method up to date in the doc? I've performed a GBS-C AIO build, including the clock mod. The clock mod was done according to both Rama's Github documentation and the GBS-C AIO documentation. But the clock generator test is not passing, for me. 

The documentation states:

> To test the clock generator is working, open the GBS Control web gui. Go to Preferences and scroll down to Activate FrameTime lock. Press on the FrameTime Lock button and the console will read "Active FrameTime Lock not necessary with external clock gen installed". Video output should be pristine and free of horizontal tearing when high output resolution presets are used.

When I perform this test, the console reads:
```
user command 5 at settings source 1, custom slot 65, status 4
FTL on
Active FrameTime Lock enabled, adjusting external clock gen frequency
```
This indicates the clock mod isn't working, but I've tried so many things to fix it. Has anyone confirmed this is up-to-date and indeed the right way to validate the clock mod?

## Things I've tried

1. Replacing my Si5351 Clock Generator PCB
2. Since the documentation stated 22 uF for the C47/C48 cap replacement, I replacing the 13 uF capacitor that came from GBS-C AIO folks with a 22 uF that measured correctly
3. Manually soldering the SCL and SDA pins directly from the associated pins on the clock board to (a) the MTV230 Myson controller pins 25 and 26 and (b) the SCL and SDA pads on top of the AIO PCB.
4. Verified all said connections with a continuity tester.

YoImLee, 2023-07-05:
Yes, "Active FrameTime Lock enabled, adjusting external clock gen frequency" means your clock generator is working correctly in more recent versions of the firmware.
 
It's also better to leave it enabled now despite having one installed to take advantage of the new and improved clockgen-based sync lock method.

brandon-arnold, 2023-07-05:
> Yes, "Active FrameTime Lock enabled, adjusting external clock gen frequency" means your clock generator is working correctly in more recent versions of the firmware.
>
> It's also better to leave it [...]

Thanks, @YoImLee . Does it do anything different if  the clock gen mod was not done correctly? 

In my troubleshooting, there were times I was pretty sure the a pin or two wasn't connected right, but as I recall I got the same output in the console.

YoImLee, 2023-07-05:
If the clock generator isn't installed (or correctly installed) and you activate the frametime lock option you will see "Active FrameTime Lock enabled, disable if display unstable or stays blank!" instead. 

This also happens if you toggle the "Disable External Clock Generator" option and restart the GBS. I believe it's also possible to get a "GBS board not responding!" message if the SCL/SDA wires are swapped around, but I'm not sure.

closed 2023-07-05 by brandon-arnold