ramapcsx2/gbs-control #477: Green Screen from ps2 when component is plugged in

ramunegaming, 2023-07-08:
Hello, I was wondering if someone could help.

I recently bought this device to be able to play my ps2 on my capture card however I can only get green to display when component is connected. I've changed the video output in the ps2 system menu so I am at a loss. Maybe I misunderstanding but does this device allow component in to HDMI out? I feel like this issue would be solved with a RGB cable but those are quite hard to get where I am living so I'd prefer to get my component cable working with this device. 

Any help would be appreciated.

invaderlex, 2023-11-23:
No, this devices allows component into upscaled VGAHV out. You need an extra transcoder VGA to HDMI. The setting on your PS2 must be YPbPr out.

Saf98, 2023-11-27:
Try nudging with a stick the capacitors c33 or c35 area. See if the image changes could be a short or bad solder.