ramapcsx2/gbs-control #481: V4 2022,  have the LDO oscillation problem? It need to fix?

h0sp1t4l3t, 2023-07-19:
Hi, 
first of all, thank you so much for make this project possible.
I'm reading the documentation, but my GBS-8200 V4 2022, is different (that zone at least) than the pics.
I have to make this fix too? 'A better fix would be to replace this capacitor with an electrolytic of 22uF to 47uF.'

Thanks!

[image]

ramapcsx2, 2023-07-20:
It looks fine. You seem to have the standard board, which is good.

evilgenius31, 2024-02-10:
Hi Rama,

Do you know if there are any differences for the v5 boards? I have a v5.2, and I want to make sure I did everything correctly: 
[image]