ramapcsx2/gbs-control #482: please help with compillation errors

Oleksandr2din, 2023-07-21:
Hello world, I am a newbie in Arduino, doing all in order with instructions I facing a lot of errors:

In file included from C:\Users\ITB\Desktop\gbs-control\gbs-control.ino:58:0:

tv5725.h:21:63: error: template argument 2 is invalid

 class TV5725 : public tw::SegmentedSlave<Addr, detail::TVAttrs>

                                                               ^

tv5725.h:25:53: error: template argument 2 is invalid

     typedef tw::SegmentedSlave<Addr, detail::TVAttrs> Base;

                                                     ^

tv5725.h:26:20: error: 'Base' is not a class, namespace, or enumeration

     using typename Base::SegValue;

                    ^

tv5725.h:27:15: error: 'SegValue' has not been declared

     template <SegValue Seg, uint8_t ByteOffset, uint8_t BitOffset, uint8_t BitWidth, tw::Signage Signed>

               ^

tv5725.h:28:31: error: 'Base' is not a class, namespace, or enumeration

     using Register = typename Base::template Register<Seg, ByteOffset, BitOffset, BitWidth, Signed>;

                               ^

tv5725.h:28:54: error: expected template-name before '<' token

     using Register = typename Base::template Register<Seg, ByteOffset, BitOffset, BitWidth, Signed>;

                                                      ^

tv5725.h:28:54: error: expected identifier before '<' token

tv5725.h:29:15: error: 'SegValue' has not been declared

     template <SegValue Seg, uint8_t ByteOffset, uint8_t BitOffset, uint8_t BitWidth>

               ^

tv5725.h:30:18: error: expected type-specifier before 'Register'

     using UReg = Register<Seg, ByteOffset, BitOffset, BitWidth, tw::Signage::UNSIGNED>;

                  ^

and much more, hopefully waiting for someone's help :) peace

nyanpasu64, 2023-07-23:
There's some possible reasons it won't build.

- Did you accidentally change the code before trying to build it? Run `git status` in your gbs-control folder and post the resulting printout here.
- What Arduino IDE version are you on? Can you screenshot your IDE window?
- What ESP8266 board software, and Tools > Board, are you using? Have you tried "LOLIN(WEMOS) D1 R2 & mini" (according to https://ramapcsx2.github.io/gbs-control/Wiki/Software-Setup.html)?

Oleksandr2din, 2023-08-04:
> There's some possible reasons it won't build.
>
> * Did you accidentally change the code before trying to build it? Run `git status` in your gbs-control folder and post the resulting printout here.
> * [...]

Hey, thank you very much for reply, but how I wrote before - I am was doing all in order with Rama's manual.

Seems like I have add some libraries to head file, which was called from another header files, I should just download they instead...

I take another computer and all starts to go )))

Oleksandr2din, 2023-08-04:
So, there is another issues, but this is another story, peace.