ramapcsx2/gbs-control #483: How to update on GBS-C ?

Yukinooooooo, 2023-07-22:
I can't update my GBS-C with ESP8266 but apparently my GBS-C is not recognised on my PC and also the serial COM port is [not listed](https://send.vis.ee/download/36491f0bdfe41285/#bkXW-FCDTgJ2_hBLRbV2OQ), I don't know why. How can I do ?

The [error message](https://privatebin.net/?ba315f11d9a9cc8c#4q2M8hzzJ1YfNHabypA2AHc2oD2qiV783fad6zCGMjHY) provided by Arduino

I've tried lots of micro usb cables but it's a failure. It's impossible to get COM for the serial port. I've even tried using other PCs, it's still a failure

but I did follow his instructions on [Software Setup](https://ramapcsx2.github.io/gbs-control/Wiki/Software-Setup.html)

ramapcsx2, 2023-07-22:
Oki, to help you single out the problem, it is definitely the OS and the USB cable / adapter. It might be simply lacking a driver, or it is defective. Get this adapter cable working first and then continue with connecting to the GBS.

YoImLee, 2023-07-22:
Looks to me like you're running into permission issues. Try running `ls -l /dev/ttyUSB0` in a terminal to see what group it belongs to, then adding your Linux user to the group, usually dialout or uucp depending on your distro. `sudo usermod -aG (groupname) (username)`

Yukinooooooo, 2023-07-22:
> Looks to me like you're running into permission issues. Try running `ls -l /dev/ttyUSB0` (with the ESP plugged in) in a terminal to see what group it belongs to, then adding your Linux user to the group, usually dialout or uucp depending on your distro. `sudo usermod -aG (groupname) (username)`
> 
> You'll need to log out or reboot for it to take effect.

The result is `crw-rw----. 1 root dialout 188, 0 Jul 22 00:30 **/dev/ttyUSB0**` from `ls -l /dev/ttyUSB0`

I also did `sudo usermod -aG mygroupname myusername `then I restarted my PC but that didn't change anything, the problem is still the same

Yukinooooooo, 2023-07-22:
> Oki, to help you single out the problem, it is definitely the OS and the USB cable / adapter. It might be simply lacking a driver, or it is defective. Get this adapter cable working first and then continue with connecting to the GBS.

However, VCP Drivers or CP210x is already installed on [Linux](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads). 

My [Micro USB cable ](https://www.amazon.com/KabelDirekt-C%C3%A2ble-Micro-charge-synchro/dp/B00L5G2GSW/ref=sr_1_1?__mk_fr_FR=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=cable+micro+usb+kabeldirekt&sr=8-1)supports Sync & Charge

Do you think my new GBS-C is defective? Could it be? It is recognised on my PC with the serial USB port, not COM.

Yukinooooooo, 2023-07-22:
> > Oki, to help you single out the problem, it is definitely the OS and the USB cable / adapter. It might be simply lacking a driver, or it is defective. Get this adapter cable working first and then continue with connecting to the GBS.
> 
> However, VCP Drivers or CP210x is already installed on [Linux](https://www.silabs.com/developers/usb-to-uart-bridge-vcp-drivers?tab=downloads).
> 
> My [Micro USB cable ](https://www.amazon.com/KabelDirekt-C%C3%A2ble-Micro-charge-synchro/dp/B00L5G2GSW/ref=sr_1_1?__mk_fr_FR=%C3%85M%C3%85%C5%BD%C3%95%C3%91&keywords=cable+micro+usb+kabeldirekt&sr=8-1)supports Sync & Charge, Plus, I bought it new.
> 
> Do you think my new GBS-C is defective? Could it be? It is recognised on my PC with the serial USB port, not COM.

YoImLee, 2023-07-22:
The result of the first command shows that you want to add your user to the dialout group so the command would be `sudo usermod -aG dialout yourusername`. 

The fact that your device shows up as tty instead of com is normal on linux.

Yukinooooooo, 2023-07-22:
> 

It works better than before but I have another [problem](https://privatebin.net/?5331e479a88e0c48#5Fuci4QE1K2cvw4GU27dmDQmXRypQYsy3M2n2gbNydoq) after connecting....

It doesn't always show the COM serial port but the new serial port is S4

[image]

YoImLee, 2023-07-22:
What happens when you select ttyUSB0 instead?

Yukinooooooo, 2023-07-22:
> 

You really are a genius, I'm not exaggerating. You really saved me. Thanks

I have another question if you don't mind, how do you connect the GBS-Control Wi-Fi to get the GBS-C interface after doing the Hard Resetting via the RTS pin...?

Yukinooooooo, 2023-07-22:
> What happens when you select ttyUSB0 instead?

You really are a genius, I'm not exaggerating. You really saved me. Thanks

I have another question if you don't mind, how do you connect the GBS-Control Wi-Fi to get the GBS-C interface after doing the Hard Resetting via the RTS pin...?

YoImLee, 2023-07-23:
After the first succesful flash your ESP should be in access point mode. You should be able to connect with for example a phone using the default password `qqqqqqqq`. Open up the browser on your phone and go to http://gbscontrol.local Then you can optionally connect your ESP to your main wifi from the GBS-C interface.

Yukinooooooo, 2023-07-23:
> 
Now it's working. Thanks again for your help

nyanpasu64, 2023-07-23:
Closing as completed. If you have any further questions, send a reply to have it reopened.

closed 2023-07-23 by nyanpasu64