ramapcsx2/gbs-control #485: YPBPR Question

ZeFoxii, 2023-08-02:
I don't have any way to test this as I just moved and don't have my setup, but a thought came to me. Let's say I want to use a wii2hdmi adapter and chose to output YPBPR and use the wii adapter as a way to get easy cheap HDMI. Would YPBPR still easily switch from 240p to 480p like it can currently. Would there by any benefits or drawbacks to my idea?

ramapcsx2, 2023-08-03:
Gee, that's a really indepth question. I *think* it should be the same as RGB out, but you'd really have to check and verify :/

ZeFoxii, 2023-08-03:
Well once I have everything built I'll be sure to let you know! I'm curious as well, I know resident evil is a big problem and chrono trigger any others?

ramapcsx2, 2023-08-04:
The problem is generally 240p/480i switching, and so any title that does those will show it, including every time you start the console (the Sony splash is in 480i).