ramapcsx2/gbs-control #486: CRT-Blur Transparency/Color blending options

ZeFoxii, 2023-08-12:
I'm not sure if this was asked before, so I apologize if there is any repetition, is it possible to add a CRT-Blur/Color-Blending effect to get something similar like natural CRT transparency/color blending, my examples of this is the standard Sonic waterfall, Dracula and Mario Kart. 

[image]
[image]
[image]
[image]

ZeFoxii, 2023-08-13:
One way I could see this being implemented is, first upscaling the image larger then the displayed resolution, then shrinking the image and displaying the image over more pixels, I feel this in conjunction with left right color blending, would give the perfect CRT-Effect.

ZeFoxii, 2023-08-13:
I could also see adding a CRT mask that cuts through the pixels that is drawn over the image.

invaderlex, 2023-09-22:
Buy a retrothink pro. This is only a scaler, a small thing enter a larger thing comes out, this is not an image processor.

AlfredoDore, 2025-06-20:
I researched and tested (of course, I don't have a GBS-C yet, I'm going to buy one) these masks are made by the signal, I tested it on my Wii with a composite and component cable, the conclusion is that the signals that give these effects together with scanlines that give definition to the image, I believe that by putting the GBS on the analog inputs of an LCD, most of these problems are solved, in my case I have an LCD monitor with VF, RCA, S-Video, Component, VGA and HDMI, an addendum, I researched with AI to make the LCD more similar to a CRT, what happens is that the sharpness level blows out the pixels, I reset the sharpness level and it worked, I also made adjustments to the image (brightness, contrast and color) I really liked the result of my image

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]