ramapcsx2/gbs-control #493: Randon flicker 2023 (jumpy image)

guibao, 2023-09-02:
I recently purchased a GBS-C, v1.7, the blue one. It still has the randon flicker problem, the screen jumps every few seconds. 
The firmware was updated to the lastet version, and dispite the fact some people stating newer versions don´t have the problem, it is still happening.

My question is: for this 1.7 version, are C33 and C35 the capacitors I must change in order to fix the issue?
Thank you!

Moogzx, 2023-11-08:
Hello, did you resolve this issue? I have opened a thread and it sounds like i may have the same problem. I also purchased the blue one (AliExpress).

https://github.com/ramapcsx2/gbs-control/issues/510

caiosilva1993, 2023-11-09:
Same here, but only with the Wii.

Metaln00b, 2023-11-12:
Have you checked if it works better without ActiveFrameTime Lock? Because I have the problem especially in higher mode (I think with X+VSST). After disabling, it was better.

Please try disabling it an reply, maybe it will help the developers. 😋

I use the orange one with OLED and RGB on a PS2 50xxx. Monitor is set to AV Mode.

ktElwood, 2024-01-11:
I build my own GBS-Control, I had "Flicker" when the unshielded I2C SDA/SDCLK lines were kind of messy and coiled up on top of the GBS8200 board.