ramapcsx2/gbs-control #500: Can on board processor be removed?

luisfcorreia, 2023-09-15:
I'm wondering if old controller could be removed, as we are already controllong the board

Curiosity, that's all :)

ramapcsx2, 2023-09-16:
Yep, in theory and a small praxis test that I did once.. it's fine :p

luisfcorreia, 2023-09-16:
additionally, I guess that the 24C12 EEPROM is also unused, or it is?

ramapcsx2, 2023-09-16:
I forgot, there's an EEPROM? So yea, if this is for data for the original MCU, off it goes! :p

luisfcorreia, 2023-09-16:
it's a small I2C connected soic chip

it will also go :)

luisfcorreia, 2023-09-16:
Apparently, you can remove all these components, depending on your usecase

enjoy hacking!

thanks for such a nice ork on a crucial project
[image]

luisfcorreia, 2023-09-16:
@ramapcsx2 Feel free to add this picture to the wiki

closed 2023-09-16 by luisfcorreia