ramapcsx2/gbs-control #502: RGB-HV output

oziphantom, 2023-09-30:
Is it possible to get RGBHV output and not RGBs output? 

Basically I'm looking to get VGA -> RGBHV which it seems nothing does.

Nuck-TH, 2023-10-04:
VGA is RGBHV by definition. Or you mean VGA 15khz?

ramapcsx2, 2023-10-04:
The default output is RGBHV already?

oziphantom, 2023-10-04:
Say the 640x480 at 60hz or possibly SVGA spec of 800x600 at 60hz transmitted via a DE-15 connector to analouge RGB 576i 0.7 p-p with a 15khz Horizontal rate at 50hz vertical or 480i at 60hz vertical suitable for a DB9 pin RGB with a H and V sync output.

ramapcsx2, 2023-10-04:
I guess you mean you want to use the bypass function, but with H+V syncs. That's something the chip can't do, or I haven't found out how to do it.

Nuck-TH, 2023-10-05:
Also this chip can't do framerate conversion.

oziphantom, 2023-10-07:
seems "GBS-Control" and this "GBSC"  https://www.ebay.com.au/itm/314802317450 are not the same thing, hence my confusion. 
Thanks for your help.

closed 2023-10-07 by oziphantom

ramapcsx2, 2023-10-07:
Well, just as info: This *is the same*, though they added a bunch of hardware to make it more convenient.
It still runs this software without any modifications (as far as I know) :)