ramapcsx2/gbs-control #509: There's a ton of people with the same issues, and nobody seems to want to help.

Murderface1982, 2023-11-05:
I've sifted through countless issue threads, can those of us who solved this issue please just give a clear concise explanation of what exactly is wrong with the GUI when it is unresponsive?  I cannot access any resolutions on the web interface, I can't change much of anything besides the toggle switches.   Please, thank you.

israpps, 2023-11-05:
I've been using my GBS Control without issues for a year.

U updated the firmware about 4 months ago and still 0 issues with the UI

taylor-madeak, 2023-11-06:
To put it concisely: GBS-C is too busy _doing its job_ (upscaling a video signal) and doesn't have any available overhead for the web UI.

This one puzzled me at first, too. But once I noticed that the UI became responsive again after I turned off the incoming video signal, it immediately made sense to me. These devices aren't exactly supercomputers, they have limitations.

Murderface1982, 2023-11-06:
> To put it concisely: GBS-C is too busy _doing its job_ (upscaling a video signal) and doesn't have any available overhead for the web UI.
>
> This one puzzled me at first, too. But once I noticed that [...]

I thought possibly the exact same thing, so i tried it with no signal, my PC was turned off and so was my TV, I'm still getting a unresponsive UI.

YoImLee, 2023-11-06:
> To put it concisely: GBS-C is too busy _doing its job_ (upscaling a video signal) and doesn't have any available overhead for the web UI.
>
> This one puzzled me at first, too. But once I noticed that [...]

This is usually because of wifi interference between the ESP and the GBS. See here: https://ramapcsx2.github.io/gbs-control/Wiki/various-notes.html

As for none of the UI buttons doing anything, this happens sometimes on a fresh install. In my experience saving a preset can make the web UI responsive again.

invaderlex, 2023-11-09:
because a ton of people fixed their wifi problem moving stuff. Where is your ESP mounted?

4lexei, 2023-11-14:
It's a bit unfair to say that we have no intention of helping. It's a free project and donations to Rama are voluntary, even he doesn't get much for his hard work.

The discussion about wifi connectivity issues have been discussed from the beginning. The solutions  have always been documented: change the heatsink, use other network channels or boost the antenna. What worked best for me was to design a box in 3D and place the ESP on the right side. Previously I used to install it vertically as Voultar showed in his video. But for me, in my particular case, it works better as I show in my photo

[image]

ramapcsx2, 2023-11-14:
Yep, it can be very small changes like that. WiFi wavelength is small, and all it takes is to remove the ESP antenna from a particular set of interference.

invaderlex, 2023-11-23:
> It's a bit unfair to say that we have no intention of helping. It's a free project and donations to Rama are voluntary, even he doesn't get much for his hard work.
>
> The discussion about wifi [...]

YES! mounting the ESP over the GBS is like mounting an antenna over a big sheet of cooper. The solder mask becomes a shield. Mine is also on the same spot as your and I have no connection problems.

ktElwood, 2024-01-11:
Try to access your ESP32 while there is no video input on the GBS82X0 or the GBS8200 itself isn't powered

I use an old phone to connect to the GBS-Control Wifi  in AP mode.

Once it's connected I have no issues using the GUI on the websever while my Wii/Ps1/Ps2  etc. are running through the GBS8200.

Lewinsky123, 2025-05-27:
Has anyone compiled the sketch to work off an ESP32-S2-WROVER, or any chip that has more power to it?

Kwakx, 2025-12-06:
@ramapcsx2 I started by updating all libraries and the platform to the latest version, and even then I noticed an improvement in the performance of the UI/websocket, and in my case, the Wi-Fi worked until the next or random restart.

Adding a few delays, status clears, etc., along with forced 802.11g mode, brought the final result. Auto reconnect is more of an extra feature.

Additionally, the update to Arduino core v3 required a change in the approach to saving credentials and requires the use of WiFi.persistent( ). 

https://arduino-esp8266.readthedocs.io/en/latest/esp8266wifi/generic-class.html
https://github.com/esp8266/Arduino/pull/7902

ramapcsx2, 2025-12-06:
Ok, sounds good. Some core driver changes in the past years might very well have fixed it :)