ramapcsx2/gbs-control #510: Faulty unit diagnosis (jagged screen)

Moogzx, 2023-11-08:
Hello. I am getting screen flickering/jagged screen with a GBS-C I have purchased from AliExpress. It starts off fine but gets progressively worse, sometimes there is no input as well.

Here are some videos of what it looks like:
1. https://drive.google.com/file/d/1mxi4Kt22MOD-SNG_PyCVBe5hdyUHCRmv/view?usp=drive_link (initial boot up)
2. https://drive.google.com/file/d/1-fwgjT3wokYEzrLW82JLoWndyTDhUP0w/view?usp=drive_link (this is after 5 minutes)

These are the settings and how it functions:
1. https://drive.google.com/file/d/15uei-W-lwUMlsSKPywueA2bM6-7pJMAB/view?usp=drive_link
2. https://drive.google.com/file/d/1tV7hKtx7FPI8bh1QQFLwVIJg-m1kLfO2/view?usp=drive_link

I have ruled out the Component cable, PS2, HDMI cable, Monitor. Only thing I haven't ruled out is the power supply to the GBSC, which I believe is fine because it is the standard requirements, and also new (5V 2A) however I have ordered another one to make sure. 

I have flashed the firmware multiple times and the same effect more or less occurs on other devices (PS3).

Does anyone know what the issue may be? I am communicating with the seller however its becoming rather difficult to diagnose the issue, as far as i can tell, I may have a faulty unit. Thanks.

Moogzx, 2023-11-09:
I have fixed the issue. Faulty power adapter. I will leave this up for anyone facing the same problem.

Saf98, 2023-11-27:
> I have fixed the issue. Faulty power adapter. I will leave this up for anyone facing the same problem.

What power supply fixed the issue? Can you post the link/specs?

Saf98, 2023-11-28:
> 5V2A, 2.1 x 5.5mm PSU
> 
> > > I have fixed the issue. Faulty power adapter. I will leave this up for anyone facing the same problem.
> > 
> > What power supply fixed the issue? Can you post the link/specs?
> 
> 5V2A, 2.1 x 5.5mm PSU

Ok my psu is 5v 3a and has issues like in the last link. The fix is replacing SoG caps but this is quite advanced.

Saf98, 2023-12-02:
> I have fixed the issue. Faulty power adapter. I will leave this up for anyone facing the same problem.

If you still have the faulty psu, can you test it again just to be sure it's not a coincidence?

Saf98, 2023-12-16:
> > > I have fixed the issue. Faulty power adapter. I will leave this up for anyone facing the same problem.
> > 
> > If you still have the faulty psu, can you test it again just to be sure it's not a coincidence?
> 
> The faulty adapter was this one:
> https://www.amazon.com.au/dp/B0C16PS3D7?psc=1&ref=ppx_yo2ov_dt_b_product_details
> 
> The one that currently works is this one:
> https://www.ebay.com.au/itm/295517585138
> 
> I suspect the first has a slight variation in voltage that causes the error. I'm not an expert on the topic, so I am unsure. I am convinced this was the cause of the issue, after process of elimination i.e (TV, HDMI cable, PS2, settings etc..)

I bought a prebuilt kaico and it has the same issues, psu didn't make a difference. But I plugged in my scart RGC Cable and it worked.
So it could be my component cables are really bad. The sync stripper could have also solved the issue for scart. I will update here if a couple quality component cable will help.