ramapcsx2/gbs-control #511: No HDMI out via SCART on PS2slim from my Kaico GBSC

Megabeans, 2023-11-09:
Hi, my brand new Kaico GBSC just arrived but I have no image at all.
My current setup is PS2 -> Multi-out to SCART -> Kaico GBSC - > HDMI -> Elgato Capture Card.
I've really tried doing research for hours, to see if I can get it working, I've managed to connect it to my wifi so now I can easily access controls, however, I've not been able to find any guide on which settings to select for my particular setup.
Can I get any help please?

Frasbo2024, 2023-12-15:
> Hi, my brand new Kaico GBSC just arrived but I have no image at all. My current setup is PS2 -> Multi-out to SCART -> Kaico GBSC - > HDMI -> Elgato Capture Card. I've really tried doing research for [...]

Did you get this working? Got exactly same issue...

Megabeans, 2023-12-15:
> Did you get this working? Got exactly same issue...

Nope. Sorry. I ended up asking Kaico for help for a whole month and they ended up asking me to send it back for a refund.

Frasbo2024, 2023-12-15:
Yeah Ive reached out to them.. but said they will get back to me on Monday.. spent hours on it today trying various settings... looks like I have a bad board...

Thanks getting back to me.

Megabeans, 2023-12-15:
> Yeah Ive reached out to them.. but said they will get back to me on Monday.. spent hours on it today trying various settings... looks like I have a bad board...
> 
> Thanks getting back to me.

No worries! It's unfortunate that it didn't work. I really wanted a good quality output for my PS2, guess I'll have to buy one of these https://kaicolabs.com/product/sony-playstation-12-hdmi-display-adapter/

Frasbo2024, 2023-12-15:
Yeah thats prob best solution for you. I have various consoles so was looking at the cheapest way to get best picture.

You know I wonder if I actually have your device thst you sent back and they have just sent me that one.. dates would make sense and they only had 2 in stock..

Megabeans, 2023-12-15:
> Yeah thats prob best solution for you. I have various consoles so was looking at the cheapest way to get best picture.
> 
> You know I wonder if I actually have your device thst you sent back and they have just sent me that one.. dates would make sense and they only had 2 in stock..

Yeah that is strange. I have the Royal Mail tracking number since I sent it back myself, it got delivered on the 14th so maybe it wasn't mine?
Anyways, I wish you the best of luck, sounds like you'll have to spend quite a bit of money to get a decent image xD OSSCs are supposedly the best of the best.

Saf98, 2024-04-09:
What cables are you using? the scart needs to be fully shielded.

Martyn575, 2024-04-17:
I've tried connecting a ps2 using component cables directly to the GBS-control and i've tried every button, every switch, every thing i can think of i can't get it working. And the only help is like one wikipage that's like about the hardware and the firmware, and 3-4 videos of people demonstrating it working, 2-3 videos of people soldering it, and a load of confused redditors. 

If i could just work out what the heck the settings are to get it to work with anything at all, like literally anything on the screen for ps2, i'd rejoice make a youtube video with the settings and share it to the world. Unfortunately, the lack of documentation has so far prohibited me from even getting the point where it's working.

Saf98, 2024-05-13:
can you send a video please?, i used ps2 preset on mine but i also updated to new firmware. Im not sure if it comes preloaded with the ps2 preset.