ramapcsx2/gbs-control #512: Color Correction settings are not saved/restored in a preset

antaenc, 2023-11-19:
The new color correction presets really help with a project I'm using GBS Control on. They don't appear to persist across power up/down  when saving a preset after adjusting the color correction settings. Should they save in the preset and be restored upon next power up/down?

YoImLee, 2023-11-21:
Seems to work fine for me, using the latest version currently available (a522b49). Are you able to save and load other settings like positioning, scaling, brightness gain and borders to your presets like normal? You may have to double-check your Arduino IDE settings and flash your ESP module again.

antaenc, 2023-11-28:
Yes a reflash and it started working as expected, thanks.

closed 2023-11-28 by antaenc