ramapcsx2/gbs-control #518: PS2 Europe - No HDMI Video but getting audio fine

ste23droid, 2023-12-27:
Hi there,
I’ve bought a GBS control pre assembled on Ali Express.
My config is this one:
PS2 Pal -> scart -> gbs control -> hdmi -> European TV.

I get audio (ps2 startup sounds) but no video.
If I enable the debug view however, I see a grey squared image on screen. If I disable it, the screen goes completely black. So the video output should be fine, I get that image.
I’ve already enabled:
- force pal 50 to 60hz
- Disable external clock generator
Based on other topics I found on issues page.

I’ve tried with multiple TVs without luck.
Can you help me out?
How can I get ps2 out properly?

I can provide developer support information of course and even logs extracted from the debug view.
Have a nice day
[image]

ramapcsx2, 2023-12-27:
Hey ho,
if you check the debug logs, is there constant "chatter" about sync being unstable and the likes?

ste23droid, 2023-12-27:
Hi there, thanks for the quick reply.

I've enabled them but cannot see at first glance if there is an issue.

I attach the debug log, if I enable "Print Info", lots of logs start to appear.
In the logs I've tried different settings but still I cannot understand why I get only audio.

[2023_12_27_debug_info.txt](https://github.com/ramapcsx2/gbs-control/files/13779843/2023_12_27_debug_info.txt)

ste23droid, 2023-12-27:
When the GBS is on, it shows it is outputting 1920x1080 at 60 Hz, RGB.
[image]

ramapcsx2, 2023-12-27:
Sync is stable. The problem is elsewhere. It's possible that you have the PS2 in YUV (not RGB) mode, or that all RGB video is not being routed to the ASIC via some custom hardware on this build.

ramapcsx2, 2023-12-27:
Yea, I mean, is that a Composite plug that you feed into the device? xD
(The scaler does RGB, not any of the "yellow plug" video forms.)

ste23droid, 2023-12-27:
Hi Robert, thanks for the hint, you solved it!

PS2 is in RGB mode.
I changed its cable to component and connected this way and it works for audio and video.

I attach pictures in case others want to connect ps2 on this build.

pal forced to 60hz and external clock generator disabled.

thanks for the help!
Have a nice day :)

[image]

closed 2023-12-29 by ste23droid