ramapcsx2/gbs-control #522: banding ?

Firas-Dhouibi, 2024-01-14:
sup yall
well i was using my gbs control for a while now and i did notice some kind of banding / faint vertical lines (more noticeable in person) but for some reason in 1080p they will be less noticeable (the lines will be super thin to notice).
[https://imgur.com/a/dyj2Oke]
as for the board itself, i did remove C11 and replace it with a 6v 22uf electro cap / added 10uf smd caps to the stock C23, C43, C42, C48 / remove the 3 RGB potentiometers and added 100ohm resistor across sync and ground for RGBs input ( since im using a retroscaler x2 in passthrough mode for my N64 s-video).
all my consoles ( NTSC N64, PAL wii and ps2 ) with official cables ( official n64 s-video cable / official ps3 component cable / official nintendo wii component cable).
tried different psu's but same result ( using a cheep 5v 2a psu for now )
any idea what can be the issue and how can i fix it ? thanks

ramapcsx2, 2024-01-15:
It is a common problem with scalers, where it is very difficult to prevent digital switching noise from the scaler chip or the RAM(!) from leaking into the RGB video.
This is usually counteracted by filters and especially by designing the PCB well. This step is where the GBS fails. 
The PCB has a few countermeasures, but it's not enough.
GBSC tries to configure the RAM and ADC parameters so as to avoid this, but there are many situations where it's not enough.
Fighting the banding thus becomes a game of tweaking parameters, until it might work in one setup, but it easily fails then on the next.
The true good fix is a better PCB, but that increases the price.
GBSC is good for the low price, but to be better requires a new PCB, so.. :p 

I can recommend you try playing with your options. Anything and nothing might help, including more hardware changes such as adding bypass capacitors.

Firas-Dhouibi, 2024-01-15:
require a new PCB ? brah like u said, GBSC is good for the low price and getting another pcb is not worth it since im gonna spend more of my dollars for a new one XD ( that if i was lucky to have another decent PCB :p ).
hmm adding more bypass caps, like stack more 10uf caps ? i also have some spare 47uf 0805 smd caps, u recommend them ?

Firas-Dhouibi, 2024-01-16:
no comment eh ? XD

Jad3d87, 2024-05-31:
I have almost identical noise issues. I can usually make it go away by disabling the clock generator, playing with the HTotal++ and HTotal--, or both.  I have found that the bars become most noticeable in 1080p with the clock gen on.