ramapcsx2/gbs-control #524: External ClockGen Active Framer Lock Not Saved Across Reset/Restart

PabloNewt, 2024-01-21:
https://github.com/ramapcsx2/gbs-control/pull/405 is working great for me. But it does not work across restarts or resets (which is pretty annoying) and I have to manually work around it every time. 

Having looked at the code I see it is because it relies on the external clock generator to have been detected at startup, yet the functionality needs the external clock gen to be disabled and active frame lock enabled. External clock gen detection is skipped if the setting "Disable External Clock Gen" has been set. Thus we can never keep the new functionality enabled across restarts/resets. 

@nyanpasu64  is this intentional to allow people to keep using the legacy active frame lock methods and because your new method isn't considered mainstream enough? If so it would seem better to add this as a new frame lock method instead of the current "overloading" of setting combinations.

nyanpasu64, 2024-01-21:
The new functionality is enabled when external clockgen is *enabled* (disable is unchecked). It depends on fine frequency adjustments not possible with internal clockgen alone.

PabloNewt, 2024-01-21:
Seems I initially got it working by disabling the external clock gen without restarting which leaves it detected and enabled and believed that was the way to turn it on! My mistake. I will close this issue.

PabloNewt, 2024-01-21:
Not an issue

PabloNewt, 2024-01-21:
Not an issue

closed 2024-01-21 by PabloNewt