ramapcsx2/gbs-control #525: GBS Control wired correctly yet no signal

KamiFolding, 2024-01-21:
Hello everyone! I recently acquired myself a GBS 8200 and a NodeMCU ESP8266 Wifi module. I wired it all up as shown in RetroRGB's video (including bridging the V4.0 Pin and connecting the Debug pin) and flashed the Wifi module (I can connect to the gbscontrol network but it's somewhat slow), yet i can't seem to get any signal whatsoever on my TV, either via SCART (S-video N64) or component (Wii). Is there any way I can identify the problem?

Thanks in advance.

ramapcsx2, 2024-01-21:
Hey, the other time we had someone using a SCART cable that was just Composite Video with an adapter.
Can you post a pic of your setup, so we can rule out the easy stuff? :)
I recommend trying with the Wii from then on, using Component. This is easier to get going.

KamiFolding, 2024-01-22:
Of course! Here is how it's all wired at the moment.

[image]
[image]
[image]
[image]

YoImLee, 2024-01-22:
The GBS can't process composite or S-video, only RGB and ypbpr component.

ramapcsx2, 2024-01-23:
Yep. In addition, the HDMI dongle adds one more layer that could go wrong. Ideally, you try to find some display that accepts VGA, just to verify that this works :)

KamiFolding, 2024-01-24:
I actually tried doing that a few times before, to no result. I can try again later today and update you if anything changes 👍

KamiFolding, 2024-01-24:
Hey, I actually got a signal this time! Seems like it refused to show any signal whatsoever, until I first powered on my Wii, and then plugged power into the GBS Control, which is strange. A few things to note was that I used VGA as a direct output instead of first converting it to an HDMI signal, and while it worked fine for a while, the screen eventually turned black and I lost signal again. Guess I'll try resoldering/reflowing all connections and see if that helps.

Also, for whatever reason, the gbscontrol Wifi network is really slow and not responding to when I apply a change, like when I'd want to upscale the resolution I click on doesn't react at all. Is this a common issue?