ramapcsx2/gbs-control #526: Calculating for new R34

rickdgray, 2024-01-24:
Hey, I have found that the R34 on my board is labeled as a 2.2k and measured in-circuit as a 1.85k. Searching through the issues history, it seems the 100 ohm was chosen under the assumption that R34 is a 1k. I'm not sure to what extent the final resistance can err. I have some 220 ohm resistors on hand; using two of those bring the in-circuit measured resistance down to 108. That still seems high, but I'm really not sure how far from 75 I should be. I did read rama's previous comments that using 75 directly can lead to syncing issues on some devices, and that it's better to aim high. But I'm not sure how high is appropriate. Any guidance? Thanks!

ramapcsx2, 2024-01-24:
I don't remember off-hand what R34 does, but when you measure stuff in-circuit, sometimes it'll produce totally wrong readings.

rickdgray, 2024-01-24:
Sorry, R34 is on the sync line out of the input VGA. The wiki states to put a 100 ohm in parallel to pull it down to 75 ohm when converting for a scart socket.

ramapcsx2, 2024-01-24:
Yeah, that is one of the tricky lines..
But it wouldn't matter all that much. The 100 Ohms will be the dominating component, where it almost doesn't matter what the other value is. You will still land around "100 - a bit" :)

rickdgray, 2024-01-24:
OK so slightly less than 100 is the target. I don't have a 100 on hand so I'll aim for that given what I do have.

closed 2024-01-24 by rickdgray