ramapcsx2/gbs-control #529: Noise / artifacting during motion?

HavoxYT, 2024-01-28:
Hello! I ordered a pre built GBS Control on Aliexpress and while it looks great on slow moving or static screens, I noticed some weird artifacting whenever an object moves on screen. It looks like some of the pixels get "stuck" while the ones around them change color. This behavior is consistent with all inputs and outputs on the scaler, and at all resolutions. Is this something that can be fixed through either a setting or simple modification or is my board faulty? Here is a video showing the problem: [https://www.youtube.com/watch?v=Hr_wFEWQo3c](https://www.youtube.com/watch?v=Hr_wFEWQo3c)

invaderlex, 2024-02-09:
There are fake Ms9288a ICs out there. The fake one looks horrible that is what is used on the fakes Wii2HDMI and PS2toHDMI dongle converters. The good ones have are lag free with good image. Try the VGA out with another converter.

HavoxYT, 2024-02-10:
I have tried using the VGA output directly into a monitor and had the same result

ramapcsx2, 2024-02-10:
It's motion adaptive deinterlacing, an algorithm that reconstructs areas of high motion differently than those that are static. There are little glitches to be expected. I could barely notice yours here :p

HavoxYT, 2024-02-10:
How could it be deinterlacing? I was playing in 480p through component not 480i. The same glitching also happens in 240p and 480i with Bob deinterlacing seleted. Notice the black dots all over the white gloves and socks on both knuckles and sonic in the intro of Sonic & Knuckles
[image]

ramapcsx2, 2024-02-10:
Ah, then it's a problem with the SDRAM, most likely. There should be a devel option in the web UI that changes the SDRAM clock. Try if that has any effect.

HavoxYT, 2024-02-10:
I'll try and get back to you. thanks!

HavoxYT, 2024-02-11:
Okay so I've messed with the SDRAM clock option and it definitely makes the graininess better but its still there. Is the SDRAM on my GBS board bad? or is there a way to fine tune this option to where it goes away completely. Thanks!

ramapcsx2, 2024-02-11:
I'm sorry, but this is one area where I really had to guesstimate on what the correct settings are.
Apparently whatever I chose does not work well for you board.
I wouldn't say that yours is defective, but the RAM does not behave like those that I've tuned for...
I can say that the majority of boards should work with the settings provided. In fact, I have not seen a misbehaving board with them yet... but what you should do next, it's hard to say.
I would recommend you contact the seller in any case, and see whether they are open to just replacing your board.