ramapcsx2/gbs-control #531: No Output on Debugger Console

Metaln00b, 2024-02-01:
Does anyone else have the problem that no output is visible on the Developer Console within the WEBUI?

Metaln00b, 2024-02-01:
No problems on [64b634](https://github.com/ramapcsx2/gbs-control/commit/64b6348efc42ea5f44f202929853e29e95d3effd).

@PSHarold So it seems to be a problem with the new OLED menu. Could it perhaps be due to the font.h being removed?

PSHarold, 2024-02-02:
Waiting for someone to confirm this as I don't have my gbsc for now.

ramapcsx2, 2024-02-02:
Web UI will run without a GBS ;p
But cheers for your work :)

YoImLee, 2024-02-02:
Just uploaded [5293343](https://github.com/ramapcsx2/gbs-control/commit/5293343b2c253a35e85cc5b638431cfaab3ed813) to my unit again and no problems here, running Firefox and Brave on Linux.

Metaln00b, 2024-02-02:
> Just uploaded [5293343](https://github.com/ramapcsx2/gbs-control/commit/5293343b2c253a35e85cc5b638431cfaab3ed813) to my unit again and no problems here, running Firefox and Brave on Linux.

If i clock are n Print Info, nothing happens. I've used Brave and Chrome, cleared the cache also.

Metaln00b, 2024-02-04:
```
if (ESP.getFreeHeap() > 20000) {
    webSocket.broadcastTXT(&data, 1);
} else {
    webSocket.disconnect();
}
```

Problem found. There is not enough heap available on my uC. I am running between 16000 and 19000. I changed the value in the statement to 10000 and now it is running.

The OLED menu now needs a little more heap, which is somehow logical.

How do we deal with this?

ramapcsx2, 2024-02-04:
That's weird that it's apparently only on yours? Maybe a fluke, who knows...
I think this check was there because the server could eat through all the leftover heap memory, so this was some protection.
It might not be needed anymore, dunno. Maybe try to remove it? :)